arduino/Arduino #4683: Arduino IDE on Windows 7 64bit

ostrozan, 2016-03-13:
Hi
while not letting  Arduino IDE as an administrator, give me an error occurs during compilation :

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-ar: unable to rename 'C:\Users\Lenovo\AppData\Local\Temp\build111c1fcbb0db364f0df2f17d70a5c782.tmp\core\core.a';** reason: Permission denied**

 I think it is not good to run the Arduino IDE as administrator

facchinm, 2016-03-14:
Hi @ostrozan ,
the `Lenovo` user Temp folder should be owned by the use itself, so you should have no problems if you are logged in as `Lenovo`. 
If it's true, you should fix the permissions to that folder (and something has gone horribly wrong in your installation).
Otherwise, if you are logged with another user, please report the IDE version you are using and some info about your setup, thanks!

PaulStoffregen, 2016-03-14:
If it's folder permission problem, seems strange the build process could get this far, which requires writing plenty of other files to that temporary folder.

Perhaps anti-virus or security is interfering with the filesystem operations when it believes executable libraries are being modified?

ostrozan, 2016-03-14:
I logged in as Lenovo (Administrator). Access to the Temp folder I enabled.
 folder in "Temp" during compilation created - the problem is only with the file "core.a"
The folder contains the same files at compile time as Administrator - missing only core.a
 I tried version 1.6.6, 1.6.7, 1.6.8

per1234, 2017-07-04:
The same issue was found to be caused by anti-virus software in this issue report: https://github.com/arduino/Arduino/issues/5616

Related: https://github.com/arduino/Arduino/issues/5956

spiderwebby, 2017-07-29:
I'm having the same issue. I'm not running a antivirus in an attempt to get this working.
Win7 x64, Arduino IDE 1.8.3
[blink compiling errors.txt](https://github.com/arduino/Arduino/files/1185175/blink.compiling.errors.txt)

spiderwebby, 2017-07-29:
successfully compiling with admin rights:
[blink compiling as admin.txt](https://github.com/arduino/Arduino/files/1185184/blink.compiling.as.admin.txt)

successfully compiling **without** admin rights with upgraded AVR-GCC from http://blog.zakkemble.co.uk/avr-gcc-builds/
[blink compiling with updated AVR-GCC.txt](https://github.com/arduino/Arduino/files/1185190/blink.compiling.with.updated.AVR-GCC.txt)

ferrouswheel, 2019-12-08:
I run into this issue with Windows 10.

`c:\program files (x86)\arduino\hardware\tools\avr\bin\../lib/gcc/avr/7.3.0/../../../../avr/bin/ar.exe: unable to rename 'core\core.a'; reason: Permission denied`

I have no antivirus enabled.

I tried deleting `.../appdata/Local/arduino15` as well as uninstalling and reinstalling.

Before I deleted `.../appdata/Local/arduino15`, this path would appear in the renaming error message. However I can't remember if it was for the `core/core.a` part or the `ar` executable.

Mayoogh, 2020-08-21:
Update ``Arduino AVR Boards`` from board manager to the latest version fixed the issue