arduino/Arduino #4686: Arduino 1.6.6+ cannot launch on 32 bit Windows 7 Ultimate version

oyao, 2016-03-14:
I used zipped version of Arduino 1.6.6 and 1.6.7 and extracted them to place under the root of C drive, but cannot launch either of them. When I run the auduino_debug.exe under Dos Window, I got following error:

C:\arduino-1.6.6>arduino_debug
Loading configuration...
Initializing packages...
com.fasterxml.jackson.core.JsonParseException: Unexpected character ('<' (code 60)): expected a vali
d value (number, String, array, object, 'true', 'false' or 'null')
 at [Source: java.io.FileInputStream@f97850; line: 5, column: 2]
        at com.fasterxml.jackson.core.JsonParser._constructError(JsonParser.java:1369)
        at com.fasterxml.jackson.core.base.ParserMinimalBase._reportError(ParserMinimalBase.java:532
)
        at com.fasterxml.jackson.core.base.ParserMinimalBase._reportUnexpectedChar(ParserMinimalBase
.java:453)
        at com.fasterxml.jackson.core.json.UTF8StreamJsonParser._handleUnexpectedValue(UTF8StreamJso
nParser.java:2278)
        at com.fasterxml.jackson.core.json.UTF8StreamJsonParser._nextTokenNotInObject(UTF8StreamJson
Parser.java:779)
        at com.fasterxml.jackson.core.json.UTF8StreamJsonParser.nextToken(UTF8StreamJsonParser.java:
665)
        at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:2926)
        at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:2873)
        at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2080)
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.ja
va:171)
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.ja
va:90)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:594)
        at processing.app.Base.<init>(Base.java:292)
        at processing.app.Base.guardedMain(Base.java:219)
        at processing.app.Base.main(Base.java:135)

All my older versions of Arduino till 1.6.5-r5 can start without any issue. Please help!
Thanks!

facchinm, 2016-03-14:
Hi @oyao, 
as a first try you could navigate to `c:\yourusername\AppData\Roaming ` and remove the `Arduino15` folder (if it exists).
If it doesn't help (or if the folder does not exist), enter `c:\yourusername\AppData\Local\Arduino15` and remove all `.json` and `.sig` files you'll find there, then restart the 1.6.7 (or 1.6.8) IDE
Let me know if it works!

oyao, 2016-03-14:
facchinm,
Your second solution: "enter c:\yourusername\AppData\Local\Arduino15 and remove all .json and .sig files you'll find there," seems working for me. Thanks, a lot.
The issue looks like from wemos packages, which I installed in 1.6.5-r5 successfully, but prevent the 1.6.6+ from starting. Now I 'm trying to install wemos package into 1.6.6, but failed! 
Anyway, thank you for your advice, at least all my arduino can start and after I restored the packages of wemos to c:\yourusername\AppData\Roaming\Arduino15, the wemos module can work for 1.6.5-r5 at least.
Thanks again!

sandeepmistry, 2016-03-14:
I'm re-opening this issue, the IDE should handle the invalid JSON file in a more graceful way.

sandeepmistry, 2016-03-14:
@oyao Could you please share how your were "trying to install wemos package".

oyao, 2016-03-14:
Sure, will do that!

oyao, 2016-03-15:
@sandeepmistry Basically I follow the the following link to install wemos package, just reinstall it with Arduino1.6.7. 

http://www.wemos.cc/Tutorial/get_started_in_arduino.html

Here are the steps:
1. Install Arduino 1.6.7 from the Arduino website.
2. Start Arduino and open Preferences window.
3. Enter    
   http://arduino.esp8266.com/stable/package_esp8266com_index.json
   
   into Additional Boards Manager URLs field. 
   You can add multiple URLs, separating them with commas.
4. Open Tools→Board:xxx→Boards Manager and install esp8266 by ESP8266 Community 
   (and don't forget to select your ESP8266 board from Tools > Board menu after installation).

After reinstalling esp8266 package with 1.6.7, it seems all version are working now and it seems the instructions for 1.6.5-r5 had some issue by that time. I think the ticket#4692 can be closed now.

PR #4692 (2016-03-17): Catch and report errors parsing contributed index files
Resolved #4686.

cc/ @facchinm
closed 2016-03-17 by cmaglie