arduino/Arduino #4689: [website] Store Support form page "File not found."

per1234, 2016-03-14:
The page https://store.arduino.cc/it/index.php?main_page=contact_us gives the message: "File not found." The link to this page is **Store Support form** on https://www.arduino.cc/en/Main/ContactUs. I think the form was moved to http://store-usa.arduino.cc/pages/contact-us-store-support (maybe there is a non-USA form also?) but the link on the Contact Us page was never updated.

Originally reported at: https://forum.arduino.cc/index.php?topic=385317 and http://forum.arduino.cc/index.php?topic=385562.msg2657391#msg2657391

sandeepmistry, 2016-07-18:
The https://store.arduino.cc/it/index.php?main_page=contact_us link is now working.

per1234, 2016-10-03:
@sandeepmistry I'm getting the "File not found" at that link again.

sandeepmistry, 2016-10-04:
@per1234 can you clarify which link has the 404?

All of these work for me:
 * https://store.arduino.cc/index.php?main_page=contact_us
 * https://store-usa.arduino.cc/pages/contact-us-store-support
 * https://www.arduino.cc/en/Main/ContactUs

per1234, 2016-10-04:
It's not the standard arduino.cc 404 page with the sad resistor, it's just some text that says `File not found` at the top left corner of the screen, that's it, no mention of 404 error anywhere.
The link is:
https://store.arduino.cc/it/index.php?main_page=contact_us
It's found at https://www.arduino.cc/en/Main/ContactUs, the "Store Support form" link.

facchinm, 2016-10-05:
@mastrolinux looks like some geo-ip dependant problem, cane you take a look?

per1234, 2016-10-05:
I suspected it might be something like that. I'm in the USA. The current report of the issue is http://forum.arduino.cc/index.php?topic=427478. The contact link they finally found that worked is https://store-usa.arduino.cc/pages/contact-us-store-support which I'm guessing indicates they're in the USA also.

per1234, 2017-07-04:
All Store Support links I can find on the website are working for me now.

closed 2017-07-04 by per1234