arduino/Arduino #4691: Leonardo requests too much USB current

HelenFoster, 2016-03-14:
ATmega32u4 boards claim to require 500mA with no fallback. This value is set in USBCore.h and reprogrammed each time a sketch is uploaded. The result is that the serial and HID connections fail on a 100mA USB port even though the board actually uses far less than 100mA.

It could be argued that the user circuit might use up to 500mA. However, the Leonardo bootloader and the Uno r3 only request 100mA (fortunately, since it would be much more difficult to change). Also, the board itself, and any user circuitry powered directly from the 5V line, still try to draw current even if the requested USB configuration is denied.

Possibilities I have thought of (not knowing if the latter two are practical):

* Change to 100mA to match the bootloader and Uno r3;
* Let the sketch set the USB current requirement;
* Request 500mA and fall back to 100mA, with some way for the sketch to check what happened so it can avoid powering up external circuitry (might also be worth considering for other boards).

agdl, 2016-07-13:
@HelenFoster this is due to the fact that you can use the 5V output on the board to power a circuit so since from USB the maxim absorbed current can be 500mA we set this value

HelenFoster, 2016-07-21:
Are you aware that this is inconsistent with the non-reprogrammable USB config in the bootloader and on similar boards? Or has the policy for new boards changed since then?

As I discussed in the second paragraph, if I use the 5V output on the board to power a 400mA circuit, the circuit will still try to draw 400mA even if the USB current request is denied. So the board insisting on 500mA is also incorrect for powering external circuits. Requesting 500mA with fallback to 100mA would be less incorrect than only 500mA or only 100mA.

In addition, the HID features are a major use case for ATmega32u4 boards. Thus they are more likely to have a low-current circuit and to need to be plugged into a low-current USB port. Refusing to work at 100mA hurts these boards particularly.

xulioc, 2017-03-26:
Arduino Leonardo and SAMD21 boards cannot be used for MIDI with iPad because of this 500mA requirement. Editing USBCore.h works great, but this is valid only for "advanced" users. I hope there is a way to define USB_CONFIG_POWER_MA from sketch asap.