arduino/Arduino #4694: IDE 1.6.8 not compiling on Windows 7

elhoncho1, 2016-03-15:
After a supposedly flawless installation, I get the following error on trying to compile the Blink example. This is taking place on a Parallels Desktop virtual machine on a Mac, in case it has anything to do. The PC\AppData path is indeed absent in the Arduino installation folder (not invisible either). Can someone help?
THANKS

Arduino: 1.6.8 (Windows 7), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Juan's PC\Documents\Arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10608 -build-path "C:\Users\Juan's PC\AppData\Local\Temp\58e9973bb6e2e908c6c199f5a9d39075.tmp\output" -warnings=none -prefs="board=uno" -prefs="build.path=C:\Users\Juan's PC\AppData\Local\Temp\58e9973bb6e2e908c6c199f5a9d39075.tmp\output" -prefs="target_package=arduino" -prefs="target_platform=avr" -prefs=build.warn_data_percentage=75 -verbose "C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino"
GetFileAttributesEx C:\Program Files (x86)\Arduino\PC\AppData\Local\Temp\58e9973bb6e2e908c6c199f5a9d39075.tmp\output: The system cannot find the path specified.

Error compiling for board Arduino/Genuino Uno.

elhoncho1, 2016-03-15:
It does create the path
 "C:\Users\Juan's PC\AppData\Local\Temp\58e9973bb6e2e908c6c199f5a9d39075.tmp\output" 
But then it looks for it at 
GetFileAttributesEx C:\Program Files (x86)\Arduino\PC\AppData\Local\Temp\58e9973bb6e2e908c6c199f5a9d39075.tmp\output

which of course does not exist (different location, the PC folder is nowhere to be found iin the Arduino installation folder)

elhoncho1, 2016-03-15:
In a separate installation, it creates a similar problem, related again to a space in a folder name. Looks like the installer tries to find a path within the Arduino folder containing a folder name corresponding to the second half of a name (starting after the space). Here it looks for Arduino\PC, where "PC" seems to come from the previous use of "Juan's PC". In the separate installation, it tries to look for Arduino\and, where the "and" comes from its previous creation of a path involving "Documents and Settings".

elhoncho1, 2016-03-15:
And a separate installation, this time on a PC running Windows 7, is doing the exact same thing. It did compile once, then started failing in the same manner described above. Is this IDE version 1.6.8 corrupt?

HineBlack, 2017-05-06:
I'm using IDE version 1.8.2 with same issue...here's my log:
```
Detecting libraries used...
"C:\Users\Hine\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10802 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/CMSIS/Include/" "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/Device/ATMEL/" "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino" "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" "C:\Users\Hine\AppData\Local\Temp\arduino_build_818263\sketch\Blink.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\Hine\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf  -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10802 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM  -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino Due"' "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/libsam" "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/CMSIS/Include/" "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\system/CMSIS/Device/ATMEL/" "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino" "-IC:\Users\Hine\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" "C:\Users\Hine\AppData\Local\Temp\arduino_build_818263\sketch\Blink.ino.cpp" -o "C:\Users\Hine\AppData\Local\Temp\arduino_build_818263\preproc\ctags_target_for_gcc_minus_e.cpp"
open NUL: The system cannot find the file specified.
Error compiling for board Arduino/Genuino Uno.
```
The folder "C:\Users\Hine\AppData\Local\Temp\arduino_build_818263\preproc\" do created correctly but it's empty.
BTW: My IDE works fine a few days ago, it just corrupted suddenly, tried to reinstall it but still has same issue.
Any ideas? Help please...

facchinm, 2017-05-08:
Hi @HineBlack ,
it looks like something in your system is preventing the IDE from reading/writing `NUL` file (which is a reserved filename in Windows). Did you update the operating system or are you using a very aggressive antivirus/firewall? Or maybe you have a file called `NUL` or `nul` in one of your libraries. Another strange behaviour is that you are compiling with Due as target but the last line reads `Error compiling for board Arduino/Genuino Uno.`

wendelborg, 2018-02-26:
Somehow Microsoft OneDrive had acquired a lock on my temp-folder, could be an idea to check for that.

elhoncho1, 2018-02-26:
How did you find out/remove this lock? I will soon be trying again. 
Thanks in advance,

consolacion, 2018-03-27:
I suddenly had the same error on IDE 1.9.0 and windows 10. Out of the blue.
Arduino: 1.9.0-beta (Windows 10), Board: "Arduino/Genuino Uno"

`GetFileAttributesEx C:\Users\myname\AppData\Local\Temp\arduino_cache_533440: The system cannot find the file specified.`

Though I have no idea why this suddenly happened, creating that directory "arduino_cache_533440"  seemed to solve it, at least for now

facchinm, 2018-03-28:
@consolacion not really related to the thread but anyway it is super strange. **Maybe** another program running in the background (like an antivirus) could have removed the folder at a certain point. The IDE creates that folder at startup, so right now the only way to get rid of such a problem is to restart it.
I'll add a simple check to recreate the directory if it gets deleted for some reason :wink:

Ismael342, 2018-05-31:
Hello friends
  this is my first comment in the group
   I have the same problem, and I have been looking for some time, what is the problem, I am using win XP I also use a laptop with win 10
         the compilation problem appears in win XP
   the thing is that after giving many flights between folders and files I realized that the IDE when creating txt files the addresses and other data have illegible characters, that happens to me with version 1.8.5, with version 1.6.0 not that problem happens and the IDE runs running in win XP
  it happens in different files, but it does not happen in the preferences file
         I think that's why the ctags_target_for_gcc_minus_e.cpp file is not found in the preproc folder that is created in TEMP
  I leave an image capture so you can see it

(https://drive.google.com/file/d/1-wGg-MNHTSIGh_j5CfaRpYGq3hNZ59rE/view?usp=sharing)

(https://drive.google.com/file/d/1EAR561RhV1jtmU-IyVQmtjmGV7Cxzr8r/view?usp=sharing)