arduino/Arduino #4696: 1.6.8 Arduino closes instantly if GPS-receiver is connected to any COM (1.6.7 works)

OrakelKSL, 2016-03-15:
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000087bc, pid=3932, tid=6152
#
# JRE version: Java(TM) SE Runtime Environment (8.0_66-b17) (build 1.8.0_66-b17)
# Java VM: Java HotSpot(TM) Client VM (25.66-b17 mixed mode windows-x86 )
# Problematic frame:
# C  0x000087bc
#
# Failed to write core dump.
...
  0x057bb000 JavaThread "cc.arduino.packages.discoverers.network.NetworkChecker" [_thread_blocked, id=1488, stack(0x067a0000,0x067f0000)]
=>0x057ee800 JavaThread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" [_thread_in_native, id=6152, stack(0x06710000,0x06760000)]
  0x05775400 JavaThread "AWT-Windows" daemon [_thread_in_native, id=6628, stack(0x05c20000,0x05c70000)]
...

facchinm, 2016-03-16:
Hi @OrakelKSL ,
1.6.8 ships with the new serial port listener which should perform significantly better than the previous one (less resources, more info, faster board recognition).
It looks like you are experiencing some sort of corner case with your gps receiver. 
Could you post here the GPS receiver infos as reported by Windows Device Manager (like VendorID, ProductID and so on)?
Also, it would be interesting if you could paste the output of listSerialC.exe (inside windows folder in the attached zip) from a command shell (cmd)
[liblistSerials-1.0.7.zip](https://github.com/arduino/Arduino/files/175529/liblistSerials-1.0.7.zip)

OrakelKSL, 2016-03-16:
Hello Martino,

 

Please see my answers:

Windows device Manager:

I have several GPS-Receivers. As example 2 different ublox Neo 7 are routed to com 7 and 8 via Hub and USB. See list.txt. One is enough to close Arduino.

 

Device-Manager: 

USB\VID_1546&PID_01A7\7&2bad6eb&0&2

USB\VID_1546&PID_01A7\7&2bad6eb&0&4

 

Output of listSerialC.exe : list.txt 

 

NMEA on Com7:

9600 8n1 Standard NMEA output:

 

$GPRMC,225227.00,A,4920.95079,N,01106.73472,E,0.085,,160316,,,A*72

$GPVTG,,T,,M,0.085,N,0.157,K,A*2D

$GPGGA,225227.00,4920.95079,N,01106.73472,E,1,08,1.13,347.5,M,47.0,M,,*5D

$GPGSA,A,3,27,16,26,21,10,20,08,18,,,,,2.31,1.13,2.02*06

$GPGSV,4,1,15,04,39,173,34,07,11,328,23,08,18,289,28,10,32,162,33*76

$GPGSV,4,2,15,13,04,038,12,15,07,073,21,16,68,239,23,18,49,116,33*73

$GPGSV,4,3,15,20,22,047,18,21,58,065,13,26,49,187,33,27,49,294,30*7E

$GPGSV,4,4,15,29,05,103,22,33,28,213,,39,32,162,*45

$GPGLL,4920.95079,N,01106.73472,E,225227.00,A,A*65

$GPTXT,01,01,02,u-blox ag - www.u-blox.com*50

$GPTXT,01,01,02,HW  UBX-G70xx   00070000 EFFFFFFFo*1B

$GPTXT,01,01,02,EXT CORE 1.00 (59843) Jun 27 2012 18:25:33*49

$GPTXT,01,01,02,ROM BASE 1.00 (59842) Jun 27 2012 17:43:52*57

$GPTXT,01,01,02,MOD NEO-7N-0*26

$GPTXT,01,01,02,PROTVER 14.00*1E

$GPTXT,01,01,02,ANTSUPERV=AC SD PDoS SR*20

$GPTXT,01,01,02,ANTSTATUS=OK*3B

$GPTXT,01,01,02,FIS 0xEF4015 found*2C

$GPTXT,01,01,02,LLC FFFFFFFF-FFFFFFED-FFFFFFFF-FFFFFFFF-FFFFFF69*20

$GPRMC,225228.00,A,4920.95073,N,01106.73467,E,0.029,,160316,,,A*75

$GPVTG,,T,,M,0.029,N,0.054,K,A*29

$GPGGA,225228.00,4920.95073,N,01106.73467,E,1,08,1.13,347.7,M,47.0,M,,*5E

$GPGSA,A,3,27,16,26,21,10,20,08,18,,,,,2.31,1.13,2.02*06

$GPGSV,4,1,15,04,39,173,30,07,11,328,22,08,18,289,27,10,32,162,30*7F

$GPGSV,4,2,15,13,04,038,10,15,07,073,20,16,68,239,22,18,49,116,31*73

$GPGSV,4,3,15,20,22,047,17,21,58,065,13,26,49,187,31,27,49,294,29*7B

$GPGSV,4,4,15,29,05,103,21,33,28,213,,39,32,162,*46

$GPGLL,4920.95073,N,01106.73467,E,225228.00,A,A*64

$GPRMC,225229.00,A,4920.95068,N,01106.73463,E,0.163,,160316,,,A*75

 

Hope it helps.

 

Best regards 

 

mit freundlichen Grüßen 
Manfred Detterbeck
_________________________________________________________________
Kornburger Strasse 9
D-90530 Wendelstein

Pos: N 49°20.960 E011°06.755

Adresse in Google Maps <http://maps.google.com/maps?q=N49.349294+E11.112450+(Manfred%20Detterbeck)&t=h&hl=de&z=18> 

Tel:       +49 9129 3425
Mobil:   +49 176 56895712

E-Mail privat:  <mailto:manfred.detterbeck5@freenet.de> mailto:manfred.detterbeck5@freenet.de 
E-Mail dienstlich:  <mailto:Manfred.Detterbeck@siemens.com> mailto:Manfred.Detterbeck@siemens.com 

 

COM7 | 1546:01A7 | (null) (null) (null)
COM8 | 1546:01A7 | (null) (null) (null)

facchinm, 2016-03-17:
Hi Manfred,
there seems to be no problem with your setup, the GPS receivers are missing a serial number (but it's not a problem).
Before investigating further, could you test if the same crash happens when using [this](http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-4482-BUILD-524-windows.zip) version of the IDE?

OrakelKSL, 2016-03-17:
Hi Martino,

 

i tried it (a veeeery slow download).

Result: With connected GPS: Refuses to start (closes after “Boards werden installiert”)

Without connected GPS: Starts, but if I connect one gps it closes instantly.

 

Same behavior of 1.6.8.

 

PS: Thank you for the very quick response! 

 

mit freundlichen Grüßen 
Manfred Detterbeck
_________________________________________________________________
Kornburger Strasse 9
D-90530 Wendelstein

Pos: N 49°20.960 E011°06.755

Adresse in Google Maps <http://maps.google.com/maps?q=N49.349294+E11.112450+(Manfred%20Detterbeck)&t=h&hl=de&z=18> 

Tel:       +49 9129 3425
Mobil:   +49 176 56895712

E-Mail privat:  <mailto:manfred.detterbeck5@freenet.de> mailto:manfred.detterbeck5@freenet.de 
E-Mail dienstlich:  <mailto:Manfred.Detterbeck@siemens.com> mailto:Manfred.Detterbeck@siemens.com

ArdWar, 2016-04-23:
Is there already any fix for this bug?

facchinm, 2016-05-02:
I still couldn't reproduce this bug with my GPS receiver, is this bug still happening in nightly version?

OrakelKSL, 2016-05-02:
Hello,

 

i tested it with a version about 1 week ago, the bug exists also in this version. (1.6.9 ???)

 

Today morning i started a trip to Spitzbergen and will return at  13.5. I cannot verify it now. (My laptop has no GPS receiver)

After returning home, i will test and report.

 

The GPS receivers are of ublox NEO7 type. 

 

 

mit freundlichen Grüßen 

Manfred Detterbeck

HorstBaerbel, 2016-05-14:
I have the same issue here with Windows 7 x64, a recent Java install and Arduino 1.6.9. I can start it ONCE after rebooting, but after that it initializes with the splash screen showing and the crashes.
Crash log:

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00008a2c, pid=2888, tid=5316
#
# JRE version: Java(TM) SE Runtime Environment (8.0_66-b17) (build 1.8.0_66-b17)
# Java VM: Java HotSpot(TM) Client VM (25.66-b17 mixed mode windows-x86 )
# Problematic frame:
# C  0x00008a2c
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x256f8c00):  JavaThread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" [_thread_in_native, id=5316, stack(0x26310000,0x26360000)]

siginfo: ExceptionCode=0xc0000005, ExceptionInformation=0x00000008 0x00008a2c

Registers:
EAX=0x0076f720, EBX=0x00000000, ECX=0x0076f73b, EDX=0x2635e706
ESP=0x2635e740, EBP=0x0076f1f8, ESI=0x00000302, EDI=0x2635e870
EIP=0x00008a2c, EFLAGS=0x00010206

Top of Stack: (sp=0x2635e740)
0x2635e740:   000007c8 00220410 2635e765 0000010b
0x2635e750:   2635e765 0000010b 2635e760 00000000
0x2635e760:   00000042 00000500 02068000 ff000003
0x2635e770:   48033600 41005500 45005700 20004900
0x2635e780:   4f004d00 49004200 45004c00 57002000
0x2635e790:   44004300 41004d00 45002000 37004d00
0x2635e7a0:   30003700 00005700 00000000 00000000
0x2635e7b0:   00000000 00000000 00000000 00000000 

Instructions: (pc=0x00008a2c)
0x00008a0c:   
[error occurred during error reporting (printing registers, top of stack, instructions near pc), id 0xc0000005]

Register to memory mapping:

EAX=0x0076f720 is an unknown value
EBX=0x00000000 is an unknown value
ECX=0x0076f73b is an unknown value
EDX=0x2635e706 is pointing into the stack for thread: 0x256f8c00
ESP=0x2635e740 is pointing into the stack for thread: 0x256f8c00
EBP=0x0076f1f8 is an unknown value
ESI=0x00000302 is an unknown value
EDI=0x2635e870 is pointing into the stack for thread: 0x256f8c00

Stack: [0x26310000,0x26360000],  sp=0x2635e740,  free space=313k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  0x00008a2c

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  processing.app.Platform.resolveDeviceAttachedToNative(Ljava/lang/String;)Ljava/lang/String;+0
j  processing.app.Platform.resolveDeviceByVendorIdProductId(Ljava/lang/String;Ljava/util/Map;)Ljava/util/Map;+2
j  cc.arduino.packages.discoverers.serial.SerialBoardsLister.retriggerDiscovery(Z)V+309
j  cc.arduino.packages.discoverers.serial.SerialBoardsLister.run()V+9
j  java.util.TimerThread.mainLoop()V+221
j  java.util.TimerThread.run()V+1
v  ~StubRoutines::call_stub
[... 231 lines cut ...]
java_command: processing.app.Base E:\ARDUINO\WiFiRGBTest\WiFiRGBTest.ino
java_class_path (initial): C:\Program Files (x86)\Arduino\lib\apple.jar;C:\Program Files (x86)\Arduino\lib\arduino-core.jar;C:\Program Files (x86)\Arduino\lib\batik-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-anim-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-awt-util-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-bridge-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-codec-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-css-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-dom-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-ext-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-gvt-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-parser-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-rasterizer-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-script-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-squiggle-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-svg-dom-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-svgpp-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-transcoder-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-util-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-xml-1.8.jar;C:\Program Files (x86)\Arduino\lib\bcpg-jdk15on-152.jar;C:\Program Files (x86)\Arduino\lib\bcprov-jdk15on-152.jar;C:\Program Files (x86)\Arduino\lib\commons-codec-1.7.jar;C:\Program Files (x86)\Arduino\lib\commons-compress-1.8.jar;C:\Program Files (x86)\Arduino\lib\commons-exec-1.1.jar;C:\Program Files (x86)\Arduino\lib\commons-httpclient-3.1.jar;C:\Program Files (x86)\Arduino\lib\commons-lang3-3.3.2.jar;C:\Program Files (x86)\Arduino\lib\commons-logging-1.0.4.jar;C:\Program Files (x86)\Arduino\lib\commons-net-3.3.jar;C:\Program Files (x86)\Arduino\lib\jackson-annotations-2.6.3.jar;C:\Program Files (x86)\Arduino\lib\jackson-core-2.6.3.jar;C:\Program Files (x86)\Arduino\lib\jackson-databind-2.6.3.jar;C:\Program Files (x86)\Arduino\lib\jackson-module-mrbean-2.6.3.jar;C:\Program Files (x86)\Arduino\lib\java-semver-0.8.0.jar;C:\Program Files (x86)\Arduino\lib\jmdns-3.4.1.j
Launcher Type: SUN_STANDARD

Environment Variables:
PATH=C:\ProgramData\Oracle\Java\javapath;c:\devkitPro\msys\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Tools;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolki [... cut ...]
CPU:total 4 (2 cores per cpu, 2 threads per core) family 6 model 37 stepping 2, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, aes, clmul, ht, tsc, tscinvbit

Memory: 4k page, physical 3854844k(2292004k free), swap 7707828k(5391052k free)

vm_info: Java HotSpot(TM) Client VM (25.66-b17) for windows-x86 JRE (1.8.0_66-b17), built on Oct  6 2015 19:59:47 by "java_re" with MS VC++ 10.0 (VS2010)

time: Sat May 14 12:44:27 2016
elapsed time: 2 seconds (0d 0h 0m 2s)
```
My COM ports look like this:
[image]

OrakelKSL, 2016-05-14:
Hello everybody,
i'm back from arctica and i testet nightly version of 9 May 2016 18:13:3 GMT.
Same behaviour: If GPS-Receiver is connected arduino closes some ms after start.
If arduino runs and a GPS-receiver get connected, it closes instantly after COM is established.
GPS-Receiver via Bluetooth are no problem, only USB connected type. (i tested several ublox types)

facchinm, 2016-05-19:
Hi @OrakelKSL @ArdWar @HorstBaerbel , could you test the build in #4971 and report if it works with your setups? Thx

HorstBaerbel, 2016-05-19:
Did not work for me, see https://github.com/arduino/Arduino/pull/4971#issuecomment-220300152

OrakelKSL, 2016-05-19:
Works! 2 GPS receivers connected, no crashes!

Good work!

closed 2017-01-19 by facchinm