arduino/Arduino #4698: Arduino Projects Book

sith77, 2016-03-16:
Page 64 Project 5 Mood Cue

The servo that comes with the kit has different wiring pin out than that of the projects learning look. The White control line and Red power line need to have the Ardiuno connections reversed from proper usage. 

Page 86 Project 8 Digital Hourglass 

The switch shown in the photo for parts needed is incorrect and shows an analog push button switch NOT a tilt switch.

facchinm, 2016-07-19:
@00alis ,could you check that?

00alis, 2016-07-19:
Hi @sith77, thanks for your feedback. Could you tell us the version of your Starterkit? In the book please go to the orange page before the Index, please tell us what the last sentence says.

For the Project 8 Digital Hourglass, it has already been corrected in a later version, as you can see
[image]

@agdl is going to doble check the servo wiring, do you have the white wire in the middle in your servo?

agdl, 2016-07-20:
Wiring is correct. All ours servo have the white in the middle

00alis, 2016-07-20:
@agdl in the illustration the white wire is not in the middle, is the illustration correct or not?
[image]

00alis, 2016-07-20:
We will update the illustration, thanks everybody!

agdl, 2016-07-20:
In the illustration effectively seems that white is not in the middle. However the colors are right:
- GND is black
- +5V is red
- PWM input is white

00alis, 2017-05-23:
Illustration updated in the new edition

closed 2017-06-09 by akash73