arduino/Arduino #4700: Arduino IDE 1.6.8 - Can't open device error while uploading (No issue in 1.6.7)

waiweng83, 2016-03-16:
I just updated from 1.6.7 and got this error message while uploading my sketch to UNO.

Sketch uses 1,066 bytes (3%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
avrdude: ser_open(): can't open device "\\.\COM14": Access is denied.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

FYI, I'm running on Windows 10 64bit. I don't have any issue with 1.6.7.

waiweng83, 2016-03-16:
On top of that, I'm getting this message when opening the serial monitor.

`Error opening serial port 'COM14'. (Port busy)`

PaulStoffregen, 2016-03-16:
Maybe this question will be too overly obvious, but have you tried rebooting your computer?

waiweng83, 2016-03-16:
Yes, I did reboot my computer and it's still the same. I even tried it on other computer running the same OS (Windows 10 64-bit) and the error still exists.

However, I noticed that this problem occurs intermittently and it may work for once after I've tried a few times.

steronydh, 2016-03-16:
I've also noticed this happening with 1.6.8, with hardware that was fine before.

VaclavSal, 2016-03-18:
I may be having similar issue but since it is outside IDE (1.6.8)  I am not going to bore this forum with details. 
If you set preferences to verbose on download and verify,  does the last line reads "CPU reset"?
I think the "problem" is that download (BOSSAC) does not release the COM port as far as the OS is concerned. 
Never had such issue running 1.6.6.

Unfortunately the COM  "error" messages almost never say "port in use".  
This is not that easy to troubleshot because most of the time it involves rebooting the OS or hardware reset ( in my case on Due). Slow process.

steronydh, 2016-03-18:
My hardware programs fine with Platformio, so it's looking like a problem with the Arduino software.

VaclavSal, 2016-03-22:
Just small add - observed several attempts to force the COM port to reset. 
It never did reset and compilation stopped. Cannot be duplicated , it is random. 
I do not know it is done by BOSSAC or is compiler doing it before BOSSAC.

ElektorLabs, 2016-03-23:
Observing exactly the same problem. Uploading works fine in 1.6.7 and older, but in 1.6.8 I get regularly the error `avrdude: ser_open(): can't open device "\\.\COM4": Access is denied.` Happens with different boards on different COM ports.

One maybe interesting extra: when I remove the MCU from the board and try again avrdude gives an unable to sync error before giving the Access denied error which seems to indicate that it can communicate at least a little with the serial driver/port, but I may be mistaken.

VaclavSal, 2016-03-23:
I cannot comment on 1..6.7. since I could never finish compiling due to some CPU "error". ( There is an error report on that which to my knowledge was never officially closed. But I may be mistaken). 
This "force COM port to reset" is apparently very old "feature" - I did find it on Github unfortunately  do not remember where - OF. ( I'll search for the exact message poster by compiler next time it happens.) 
I noticed that some BOSSAC now report "time of process", it is probably just passing another parameters to BOSSAC and has nothing to do with this forced reset. 
But to me it illustrates that current team is more interested in cosmetics instead of making the IDE work better. 
For example - now we have option to select from last 10 ( or so , I did not count it) used files instead of ONLY 4. Huge improvement and very difficult to code. Thanks.

ozligia, 2016-03-25:
Did some research.
My system: Windows 10, Version 1511, Build 10586. 164
Same symptoms.

The thing is that if i click quickly "Open serial monitor" sometimes it opens. But the most time result is: Error opening serial port 'COM4'. (Port busy).
But if i close Arduino IDE i connect to my board without any issue. Then i run Arduino IDE and cannot connect to my board again.

I ran  process explorer and find out, that the registry key
HKLM\HARDWARE\DEVICEMAP\SERIALCOMM is opened permanently.

f i close this handle forcibly it will be reopened immediately. But i managed to close it and click open serial monitor. When serial monitor is opened, registry key is not being held.

If i connect putty to COM before opening Arduino IDE, Arduino IDE cannot open HKLM\HARDWARE\DEVICEMAP\SERIALCOMM,

It seems some kind of bug. Maybe system requires access to this key before it can access COM port. But the key is being held by Arduino IDE permanently.

VaclavSal, 2016-03-26:
My last and final attempt to make 1.6.8 work on XP produced some java "touching COM" port errors. 
Since 1.6.6. changed Java  I am pretty  much stuck with 1.6.5. and so far had no issues with COM port.

facchinm, 2016-03-30:
Hi @ozligia, are you experiencing this problem with all your boards or with one in particular? Does it use an FTDI chip as USB-to-serial converter?
BTW, if the problem is the Registry key being held, the relevant piece of code is this one https://github.com/facchinm/libserialport/blob/658f2124144bca5ef033e0e5308c2434da27bee4/windows.c#L481
The function always executes `RegCloseKey` before returning, so it should be safe (maybe not thread-safe but I'm trying to investigate)

ElektorLabs, 2016-04-01:
arduino-PR-4792-BUILD-534 doesn't seem to work at all on my Windows 7 system. After the splash screen it dies. Any special instructions?

facchinm, 2016-04-01:
Can you launch `arduino_debug.exe` from a `cmd` shell and paste the error dump? Thanks

ElektorLabs, 2016-04-04:
Sorry for the slow reply, I was away this weekend:
`c:\work\Arduino\arduino-PR-4792-BUILD-534>arduino_debug
Loading configuration...
Initializing packages...
Preparing boards...
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x619416f6, pid=2760, tid=4004
#
# JRE version: Java(TM) SE Runtime Environment (8.0_66-b17) (build 1.8.0_66-b17)

# Java VM: Java HotSpot(TM) Client VM (25.66-b17 mixed mode windows-x86 )
# Problematic frame:
# C  [listSerialsj.dll+0x16f6]
#
# Failed to write core dump. Minidumps are not enabled by default on client vers
ions of Windows
#
# An error report file with more information is saved as:
# c:\work\Arduino\arduino-PR-4792-BUILD-534\hs_err_pid2760.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

c:\work\Arduino\arduino-PR-4792-BUILD-534>`

After using `liblistSerials-1.0.9\windows\listSerialsj.dll` the IDE starts, but crashes when Windows finds my board. Here is the crash dump:

`c:\work\Arduino\arduino-PR-4792-BUILD-534>arduino_debug
Loading configuration...
Initializing packages...
Preparing boards...
Starting...
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x74ca08f5, pid=3460, tid=3212
#
# JRE version: Java(TM) SE Runtime Environment (8.0_66-b17) (build 1.8.0_66-b17)

# Java VM: Java HotSpot(TM) Client VM (25.66-b17 mixed mode windows-x86 )
# Problematic frame:
# C  [msvcrt.dll+0x108f5]
#
# Failed to write core dump. Minidumps are not enabled by default on client vers
ions of Windows
#
# An error report file with more information is saved as:
# c:\work\Arduino\arduino-PR-4792-BUILD-534\hs_err_pid3460.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

c:\work\Arduino\arduino-PR-4792-BUILD-534>`

About my system:

`OS: Windows 7 , 64 bit Build 7601 (6.1.7601.19160)

CPU:total 4 (2 cores per cpu, 2 threads per core) family 6 model 42 stepping 7, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, aes, clmul, ht, tsc, tscinvbit

Memory: 4k page, physical 4095412k(2167560k free), swap 8188988k(5583740k free)

vm_info: Java HotSpot(TM) Client VM (25.66-b17) for windows-x86 JRE (1.8.0_66-b17), built on Oct  6 2015 19:59:47 by "java_re" with MS VC++ 10.0 (VS2010)
`

Hopes this helps.

facchinm, 2016-04-07:
Solved by #4792

closed 2016-04-07 by facchinm

YucelyBeb, 2019-03-15:
Could you solve the problem? I have the same problem and I do not know how to solve it