arduino/Arduino #4701: Blurry Serial Monitor icon when switching applications

per1234, 2016-03-16:
Arduino IDE 1.6.9 2016/03/15 03:33 with Windows 7 64 bit
- Open Serial Monitor
- Press Alt+Tab
The Arduino IDE application icon is sharp but the Serial Monitor icon is blurry:
[image]

Using Arduino IDE 1.6.6 both icons are blurry:
[image]

I suspect that the fix for https://github.com/arduino/Arduino/issues/3473 was responsible for the IDE icon becoming sharp after 1.6.6 so maybe a similar fix could be applied to the serial monitor?

facchinm, 2016-03-25:
Should be solved by https://github.com/facchinm/Arduino/commit/2c564e68b91b71871fb22e35278579eccb8f79dc (only tested on Linux, though).
Could you test the latest build produced by ArduinoBot in this page? https://github.com/arduino/Arduino/pull/4482

per1234, 2016-03-25:
Looks good on Windows 7:
[image]
Thanks facchinm!

commit a3297315 (2016-04-01): use hi-res icons for serial monitor and plotter

fixes #4701
closed 2016-04-01 by facchinm