arduino/Arduino #4702: travis ci build fails with arduino 1.6.7 and 1.6.8 and boards manager url

tzapu, 2016-03-16:
but seems to work fine with 1.6.5

1.6.5 ok
```
$ arduino -v --verbose-build --verify $PWD/examples/AutoConnect/AutoConnect.ino
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Xlib:  extension "RANDR" missing on display ":1.0".
Initializing packages...
Preparing boards...
Verifying...
Using library ESP8266WiFi in folder: /home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi 
Using library DNSServer in folder: /home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/DNSServer 
Using library ESP8266WebServer in folder: /home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WebServer 
Using library WiFiManager in folder: /usr/local/share/arduino/libraries/WiFiManager 
/home/travis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266 -I/home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/variants/generic -I/home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi/src -I/home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/DNSServer/src -I/home/travis/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WebServer/src -I/usr/local/share/arduino/libraries/WiFiManager /tmp/build661832954510790350.tmp/AutoConnect.cpp -o /tmp/build661832954510790350.tmp/AutoConnect.cpp.o 
```

1.6.7 failed
```
$ arduino -v --verbose-build --verify $PWD/examples/AutoConnect/AutoConnect.ino
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Xlib:  extension "RANDR" missing on display ":1.0".
Initializing packages...
Preparing boards...
Verifying...
/usr/local/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware "/usr/local/share/arduino/hardware" -hardware "/home/travis/.arduino15/packages" -tools "/usr/local/share/arduino/tools-builder" -tools "/usr/local/share/arduino/hardware/tools/avr" -tools "/home/travis/.arduino15/packages" -built-in-libraries "/usr/local/share/arduino/libraries" -libraries "/home/travis/Arduino/libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10607 -build-path "/tmp/buildc98e94dd602d6e73bee2cc93694b0ca7.tmp" -warnings=null -prefs="boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json" -prefs=build.warn_data_percentage=75 -verbose "/home/travis/build/tzapu/WiFiManager/examples/AutoConnect/AutoConnect.ino"
Property line '//arduino.esp8266.com/stable/package_esp8266com_index.json"' is invalid
Error compiling.
```

has the boards url param changed? or maybe it needs some escaping?

the cli params are
```
install:
  - ln -s $PWD /usr/local/share/arduino/libraries/WiFiManager
  - arduino --pref "boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json" --save-pref
  - arduino --install-boards esp8266:esp8266
  - arduino --board esp8266:esp8266:generic --save-prefs
```

plinioseniore, 2016-04-01:
I've the same issue and haven't figured out where the problem is. Any suggestion?

per1234, 2016-04-01:
Well it may not be relevant to the issue but I notice that 
```
arduino --pref "boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json" --save-pref
```
should be 
```
arduino --pref "boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json" --save-prefs
```
Do you get the same error manually running that command from the command line? That works fine for me.

plinioseniore, 2016-04-02:
In my case the syntax looks correct but I have the same problem,
https://github.com/souliss/souliss/blob/esp8266_core2/.travis.yml

sandeepmistry, 2016-04-08:
Something is definitely up, my workaround for now is to clear it afterwards:
```
arduino --pref "boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json" --install-boards esp8266:esp8266
```
```
arduino --pref "boardsmanager.additional.urls=" --save-prefs
````

I'll try to reproduce in a non-Travis CI environment, seems like something is wrong in the preference saving area, as the boards support install works.

sandeepmistry, 2016-04-12:
This is solved in the latest hourly builds by #4853. Also, https://github.com/arduino/arduino-builder/issues/132 will prevent ```arduino-builder``` from splitting on pref args that contain ```:```'s.

closed 2016-04-12 by sandeepmistry