arduino/Arduino #4707: Please add AnalogMultiButton library

dxinteractive, 2016-03-17:
I've made a library for easily reading presses from multiple buttons through a single analog input using a voltage ladder. It does debouncing, and provides a set of functions to detect when buttons are being held, pressed, released, pressed for a duration, released before or after a specific duration has elapsed, and also allows a button being held to repeatedly report a "press" on a timed interval for use with cursors or scrolling. Could I please get this added to the list of libraries? Hopefully it'll be nice and easy for most people to use given the examples:

[https://github.com/dxinteractive/AnalogMultiButton](https://github.com/dxinteractive/AnalogMultiButton)

Please let me know if you need anything fixed up first.

agdl, 2016-03-17:
@dxinteractive your library cannot be added. library.properties is missing, tag and examples too. Please follow this guide
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

and take a look on how Arduino libraries are structired

https://github.com/arduino-libraries/RTCZero

dxinteractive, 2016-03-17:
OK thanks, I looked around for a guide on how to structure it but didn't come up with anything, I'll resubmit once it's ready. Thanks again

agdl, 2016-03-17:
I linked you a library to see how they are structured

dxinteractive, 2016-03-26:
Hi agdl, I've followed the standards you've provided and think I have it ready. Changes have been pushed, could you please review?

agdl, 2016-03-29:
Sorry about the delay. However it can't be added yet since a tag is missing

dxinteractive, 2016-03-30:
Ok done, thanks

agdl, 2016-03-30:
Done thank you!

closed 2016-03-30 by agdl