arduino/Arduino #4708: Automatic forward declaration breaks when function prototype is split across two lines

stefandz, 2016-03-17:
When you have a function whose prototype immediately before its implementation is split across two lines, it seems that this breaks Arduino automatic forward declaration. Minimal example to see this:

    void setup() {
      breakArduino();
    }
    
    void loop() {
    }
    
    void breakArduino
    () {
      
    }

gives the following error at compile time:

    test:2: error: 'breakArduino' was not declared in this scope
       breakArduino();
                    ^
    exit status 1
    'breakArduino' was not declared in this scope

which can be fixed by replacing the code with

    void breakArduino();
    
    void setup() {
      breakArduino();
    }
    
    void loop() {
    }
    
    void breakArduino
    () {
      
    }

Strangely, I think this is something that used to work, but now doesn't. We have some user contributed code that must have compiled once, but suddenly I am getting a lot of support requests for it based around this.

facchinm, 2016-03-17:
Hi, the error you are experiencing has already been reported to the arduino-builder repo arduino/arduino-builder#80
A possible solution is here https://github.com/arduino/arduino-builder/pull/117 but has not been merged yet
I'm attaching the binaries for arduino-builder which should solve the problem, please test them (by replacing `arduino-builder` binary in your installation folder with the one provided suitable for your platform) and report the results in the linked issue :wink: 
[arduino-builder-pr117.zip](https://github.com/arduino/Arduino/files/177633/arduino-builder-pr117.zip)

stefandz, 2016-03-17:
Thanks Martino - I had checked the Arduino issues list for dupes, but didn't think about the builder repo. This definitely works when dropped in to Arduino 1.6.7 on OSX. Will comment in the linked issue too. Thanks, and keep up the great work!

closed 2016-03-17 by stefandz