arduino/Arduino #4710: How to retive subject files from github?

VaclavSal, 2016-03-18:
I  would not ask this HERE but run into a snag with Arduino community REUSING "library" AKA header file  name. Than BTW comments pops out from IDE (1.6.8) compiler - "it works only in AVR" 
In other words - building "libraries" with same name and different function is a horrible idea. 
(Can be easily verified by checking Arduino.cc forum) 

I would like remedy this and like to get ENTIRE  githuib repository code downloaded in zip format.

One source said there should be a ZIP button on page somewhere. 
I do not see any ZIP buttons on this particular repository.

After trying different  way I am getting "need to enable push".
 How do I "enable push"? 

I like to retrieve the code directory as ZIP and return it back as ZIP.

matthijskooijman, 2016-03-18:
I'm not exactly sure what you're asking here - do you want to download third-party libraries from github? If so, this doesn't really sound like an Arduino issue, but something to take up with github or that third-party.

As for downloading a zip file from *this* repository, there is a download zip button in the top right of the files overview:

[image]

> I like to retrieve the code directory as ZIP and return it back as ZIP. 

Not sure what you mean with "return it back", do you mean you want to make changes and submit those as a zipfile? If so, then github has no support for that - submitting changes to a project is done through Pull Requests, which should be made through some git client. Individual developers/projects might accept contributions as zip files, but since this is a lot more work and a lot harder to review/discuss than proper pullrequests, I do not expect many will accept this (I'm pretty sure the Arduino project does not).

PaulStoffregen, 2016-03-18:
Adafruit has been taking an approach of putting big green download buttons in the center of their tutorials, with plenty of white space around all 4 sides, where the button links directly to the ZIP file.  This is almost certainly because they're learned novice users are overwhelmed and confused by the large number of links and feature on Github pages.

VaclavSal, 2016-03-18:
Thanks, since this was my first attempt I did not realize I have to download the entire repository.
Just did not go far enough to see the "zip" button. 
For the other comments I take it as Arduino would rather continue to screw around and keep same names for header files with totally different functionality. Case of point "LiquidCrystal_I2C". 

I know this this has been explained many times - but it would be nice to have some simple document explaining where different "libraries" reside. For example "contributed " libraries directory is where?

Also once the "library" is loaded there in no easy way to delete it - as in my case wrong LiquidCrystal.

VaclavSal, 2016-03-22:
Could someone close this as solved? Please. 
Or can I do it myself ?

matthijskooijman, 2016-03-22:
You should be able to close your own issues with the "Close" button below the comment field. I've gone ahead and closed this one for you.

closed 2016-03-22 by matthijskooijman

VaclavSal, 2016-03-22:
Thanks, I was not aware it was legal to close stuff here. 
Just do not want to get kicked out for same reason I got kicked out from Arduio.cc - deleting my own post. .