arduino/Arduino #4712: Unable to launch Arduino IDE 1.6.8 on Mac OS X 10.11 (El Capitan)

florianb, 2016-03-18:
Hi, i'm new to Arduino and tried using the Arduino IDE. Unfortunately the IDE does not show up, it exits silently after the splash-screen. A view into the console shows the following error:

    18.03.16 09:04:38,472 com.apple.xpc.launchd[1]:
      (cc.arduino.Arduino.177312[97855]) Service exited with abnormal code: 255

As recomended i'm running

    java version "1.7.0_80"
    Java(TM) SE Runtime Environment (build 1.7.0_80-b15)

on

    System Version: OS X 10.11.3 (15D21)
    Kernel Version: Darwin 15.3.0

I'd be thankful for every hint which may cause this error.

facchinm, 2016-03-18:
Hi @florianb ,
could you try starting the IDE directly via the console (using `./Arduino.app/Contents/MacOS/Arduino`) and paste here the complete stacktrace?
Thanks!

florianb, 2016-03-18:
Hi @facchinm - of course!

:joy_cat:

If i run `./Arduino.app/Contents/MacOS/Arduino` it starts like a charm..

May it be that the Java-terminal-environment differs from the launchd-environment? :thought_balloon:

florianb, 2016-03-18:
Btw - the clicked startup fails in or after "Loading Configuration".

facchinm, 2016-03-18:
Mmmmhh quite strange, because the JRE should be self contained...
I'm leaving this on hold until we are able to reproduce the crash executing via terminal, sorry.
In the meantime we could ask some OSX ninja if he knows what is happening :smile:

florianb, 2016-03-18:
Crazy thing.. i'll think about that issue.. :hocho:

chrstphrchvz, 2016-03-20:
Another user here…I haven't come across this issue myself, but the symptoms described here are similar to those in #3365, which for some users was caused by "incompatible tools" (e.g. TechWillSaveUs gamer library), and/or a "version mismatch" in a Java library (bcprov) that was present in both in the Arduino package and a Java extensions folder on their machine--the latter of which was supposedly fixed, but at least one user has encountered since.

florianb, 2016-03-20:
@chrstphrchvz thanks a lot - i did not try to start it again via the app-icon until now. Unfortunately (or luckily?) it works -- so i'm unable to verify the bug.

However i tried to find out, if my system meets any of the suspected criterias and i guess it does not. This was the first time i installed the Arduino IDE so there weren't any additional tools installed. The `bcprov`-library exists only in the Arduino-app.

Since i cant' track the issue down i'll close this issue for now. If it reappears i'd like to come back here! Thanks for your help so far! :candy:

closed 2016-03-20 by florianb

renowden, 2019-01-02:
I am getting this issue - MacOS 10.11.6 on Mac Pro (Early 2008).

[Auduino IDE fail.txt](https://github.com/arduino/Arduino/files/2720565/Auduino.IDE.fail.txt)

Enclosed terminal session as requested above.