arduino/Arduino #4713: Sketch always rebuilded

Testato, 2016-03-18:
on IDE 1.6.8, win7_64, at every Verification in the consolle appear:
-------------
Build options changed, rebuilding all
-------------

also if you do not change nothing.
If you verify two times the Blink, whitout touch it, all the code is rebuilded and need more time than befoure

Testato, 2016-03-18:
seems that the rpoblem is present only when switch from an architecture to another. Example if you try blink on AVR, it do not rebuild, but if you try Blink on ESP and after you return to AVR, it start to rebuild every verification, also if you remain always on AVR

Testato, 2016-03-18:
Test the bug in this way:

- Open Arduino IDE, it will open on empty sketch whit setup() and loop()
- add a simply space in the loop function
- press Verify

the "Build options changed, rebuilding" will writed in the consolle and the verification process is slow.
If you try to verify a second time, the Build options changed, rebuilding always return

facchinm, 2016-03-25:
should be solved by arduino/arduino-builder#129 , please test the nightly to check if the fix behaves correctly

closed 2016-03-25 by facchinm

Testato, 2016-03-29:
it seems to work