arduino/Arduino #4714: 16u2 USB2Serial firmware does low baud rates wrong

WestfW, 2016-03-19:
http://forum.arduino.cc/index.php?topic=385567.msg2670448#msg2670448

The xxxU2 firmware for implementing USB to Serial does not handle very low bitrates (300bps) correctly.
These low bitrates are not possible on a 16MHz AVR using the U2X doublespeed setting (which is used for Arduino, normally.)   The AVR firmware contains code that checks for an overly high divisor and unsets U2X, but it does not recalculate the divisor. 

```
 /* Special case 57600 baud for compatibility with the ATmega328 bootloader. */	
	UBRR1 = (CDCInterfaceInfo->State.LineEncoding.BaudRateBPS == 57600)
			? SERIAL_UBBRVAL(CDCInterfaceInfo->State.LineEncoding.BaudRateBPS)
			: SERIAL_2X_UBBRVAL(CDCInterfaceInfo->State.LineEncoding.BaudRateBPS);	

	UCSR1C = ConfigMask;
	UCSR1A = (CDCInterfaceInfo->State.LineEncoding.BaudRateBPS == 57600) ? 0 : (1 << U2X1);
	UCSR1B = ((1 << RXCIE1) | (1 << TXEN1) | (1 << RXEN1));
```

The SAM firmware does not have any checks at all to unset U2X.

NicoHood, 2016-03-19:
Does this also apply to HoodLoader2?
https://github.com/NicoHood/HoodLoader2/blob/master/avr/bootloaders/HoodLoader2/HoodLoader2.c#L1108-L1129

Do you have a simple test sketch @WestfW ? Or even a solution?

SpenceKonde, 2016-03-19:
The solution is to set U2X the UBRR1 and U2X registers correctly for low baud rates? 

Test sketch is *trivial* - Serial.begin(300); Serial.println("test"); connect to serial monitor at 300 baud, observe gibberish. 
I think this would work:

```

 /* Special case 57600 baud for compatibility with the ATmega328 bootloader. */ 
    UBRR1 = (CDCInterfaceInfo->State.LineEncoding.BaudRateBPS == 57600 || CDCInterfaceInfo->State.LineEncoding.BaudRateBPS < 600)
            ? SERIAL_UBBRVAL(CDCInterfaceInfo->State.LineEncoding.BaudRateBPS)
            : SERIAL_2X_UBBRVAL(CDCInterfaceInfo->State.LineEncoding.BaudRateBPS);  

    UCSR1C = ConfigMask;
    UCSR1A = (CDCInterfaceInfo->State.LineEncoding.BaudRateBPS == 57600 || CDCInterfaceInfo->State.LineEncoding.BaudRateBPS < 600) ? 0 : (1 << U2X1);
    UCSR1B = ((1 << RXCIE1) | (1 << TXEN1) | (1 << RXEN1));

```

oqibidipo, 2016-03-19:
> Does this also apply to HoodLoader2?

No. I just installed it on my Uno and it works like a charm at 300.

facchinm, 2016-03-25:
Should be solved with new bootloaders too (temporarily hosted at https://github.com/facchinm/Arduino_avrusb_firmware)