arduino/Arduino #4715: Set of TinyProtocol libraries to add to Arduino Playground

lexus2k, 2016-03-19:
Set of TinyProtocol libraries are developed to encapsulate packets communication over different physical channels. Packet structure used is similar to HDLC/PPP framing (but not follow it). FCS supported are no check, simple 8-bit checksum, CRC16-CCITT, CRC32-CCITT.
Protocol can be used to communicate over UART from the box. It allows to avoid manual synchronization between 2 boards or the board and PC. There are 3 libraries provided: Nano, Micro and Full. Nano has the smallest memory consumption and supports only 8-bit checksum. Full version supports all capabilities.

https://github.com/lexus2k/tinyproto/tree/master/releases/arduino/TinyProto-Nano
https://github.com/lexus2k/tinyproto/tree/master/releases/arduino/TinyProto-Micro
https://github.com/lexus2k/tinyproto/tree/master/releases/arduino/TinyProto-Full

per1234, 2016-03-19:
The Arduino Playground is a public wiki so you can add links to your libraries to it yourself. There is a password entry page to prevent spam but the password is prefilled so anyone can edit the Playground. If you meant you wanted your libraries added to the Arduino Library Manager then you will need to make some changes, see: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ.

lexus2k, 2016-03-26:
Ok. Thank you. I think I will add my library to Playground wiki.
As for Arduino Library Manager, it is not clear from the FAQ, you mentioned, what's wrong with the structure of my libraries. Does it mean that I need to have separate repository for each version of Tiny Protocol library: Nano, Micro and Full?

per1234, 2016-03-26:
>Does it mean that I need to have separate repository for each version of Tiny Protocol library: Nano, Micro and Full?

If they're each separate libraries as is now in the links you posted, yes. If you can combine them into a single library then you could do it in one repository. Your current folder structure is not compatible with Library Manager. A way you can test this is to [Download ZIP](https://github.com/lexus2k/tinyproto/archive/master.zip) or [the release source zip file](https://github.com/lexus2k/tinyproto/archive/v0.1.2.zip) and then try to do **Sketch > Include Library > Add .ZIP Library...** with the downloaded file. If you do that with the current folder structure of the repository you get the error: `Specified folder/zip file does not contain a valid library`.

lexus2k, 2016-03-30:
Ok. I added link to Arduino Playground. Also I reworked library structure and combined all versions into single library. But I had to create separate repository for Arduino: https://github.com/lexus2k/arduino-protocol
I checked that Library Manager allows to add new library. So, now it is possible to use the library with Library Manager

closed 2016-07-18 by sandeepmistry