arduino/Arduino #4721: Missing filename error after update

Enfermeria, 2016-03-19:
After updating (it was solicited by the arduino IDE), I get this error when trying to compile:

============================================================================
Arduino:1.6.6 (Windows 10), Placa:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
avr-g++: error: missing filename after '-o'
exit status 1
Error de compilación
============================================================================

Can you help me?

Thanks in advance!!!
                                                     John

PD: If you want to see the detailed compilation messages:
============================================================================
Arduino:1.6.6 (Windows 10), Placa:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\john\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\john\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\john\Documents\Arduino\libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10606 -build-path "C:\Users\john\AppData\Local\Temp\build9a934a4b06bc355b0bbd1a523e3eaad8.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\john\Documents\Arduino\Morse01\Morse01.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\john\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\john\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\john\Documents\Arduino\libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10606 -build-path "C:\Users\john\AppData\Local\Temp\build9a934a4b06bc355b0bbd1a523e3eaad8.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\john\Documents\Arduino\Morse01\Morse01.ino"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\mega" "C:\Users\john\AppData\Local\Temp\build9a934a4b06bc355b0bbd1a523e3eaad8.tmp\sketch\Morse01.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\mega" "C:\Users\john\AppData\Local\Temp\build9a934a4b06bc355b0bbd1a523e3eaad8.tmp\sketch\Morse01.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\mega" "C:\Users\john\AppData\Local\Temp\build9a934a4b06bc355b0bbd1a523e3eaad8.tmp\sketch\Morse01.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\mega" "C:\Users\john\AppData\Local\Temp\build9a934a4b06bc355b0bbd1a523e3eaad8.tmp\sketch\Morse01.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\mega" "C:\Users\john\AppData\Local\Temp\build9a934a4b06bc355b0bbd1a523e3eaad8.tmp\sketch\Morse01.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\john\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\mega" "C:\Users\john\AppData\Local\Temp\build9a934a4b06bc355b0bbd1a523e3eaad8.tmp\sketch\Morse01.ino.cpp" -o ""
avr-g++: error: missing filename after '-o'

exit status 1
Error de compilación

per1234, 2016-03-19:
The problem is caused by using Arduino AVR Boards 1.6.10 with Arduino IDE 1.6.6, they are incompatible. The solution is to either:
- Use Arduino IDE 1.6.7 or 1.6.8.
-  Install Arduino AVR Boards 1.6.8(due to a [bug](https://github.com/arduino/Arduino/issues/4130) you can't go back to using 1.6.9) by: **Tools > Board > Boards Manager... > Arduino AVR Boards**(click on it) **> Select version > 1.6.8 > Install**. After this you will continue to get updatable notifications about the more recent Arduino AVR Boards version so you might want to **File > Preferences > Check for updates on startup**(uncheck) to disable this notification.

Enfermeria, 2016-03-21:
Thank you very much for your answer. I'll reinstall it. Thanks

closed 2016-03-21 by Enfermeria