arduino/Arduino #4723: [Reference] attachInterrupt() example and documentation corrections

per1234, 2016-03-20:
On the reference page https://www.arduino.cc/en/Reference/AttachInterrupt the example attaches the interrupt to pin 13 but that won't work for common Arduino boards such as Uno and Mega. Also I think it's better to use different pins for the interrupt and the output. I would suggest changing the interrupt pin to 2 or 3 as this will work on any of the listed boards except MKR1000 Rev.1. The updated example would look something like:
```arduino
const byte ledPin = 13;
const byte interruptPin = 2;
volatile byte state = LOW;

void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
}

void loop() {
  digitalWrite(ledPin, state);
}

void blink() {
  state = !state;
}
```

Originally reported at http://forum.arduino.cc/index.php?topic=385081.0 and https://forum.arduino.cc/index.php?topic=358930.0

A couple other typos in the `attachInterrupt()` reference page:
>Normally you should use digitalPinToInterrupt(pin), rather than place interrupt an number directly into your sketch. 

should be changed to:
>Normally you should use digitalPinToInterrupt(pin), rather than place an interrupt number directly into your sketch. 

and:
> Four contstants are predefined as valid values:

should be changed to 

> Four constants are predefined as valid values:

agdl, 2016-03-30:
Done thank you!

closed 2016-03-30 by agdl

per1234, 2016-03-31:
@agdl I don't see the changes on the reference page.

per1234, 2016-04-11:
@agdl I still don't see the changes on https://www.arduino.cc/en/Reference/AttachInterrupt. If you don't have time to deal with this right now please reopen the issue so it doesn't get lost.

facchinm, 2016-04-12:
Sorry @per1234 , the corrected version was ready but we didn't push itto production server :smile: Thanks for the follow up!

Omar-alSuntawi, 2019-03-27:
I found that even with the corrected code posted now in 2019 March 27, is not correct.