arduino/Arduino #4724: 1.6.8 IDE and 1.05 core for Arduino 101 does not install Windows driver

xinjincanada, 2016-03-20:
Document said using boards manager installing the 101 board core would automatically install windows driver, but it did not happen. Discussions in the forum pointing me to manually install the driver from following path, still does not work.

C:\Documents and Settings\user\Local Settings\Application Data\Arduino15\packages\Intel\tools\arduino101load\1.6.4+1.18\drivers  

IDE shows only COM1 available to choose from, but device manager shows on COM3 for the virtual port. Two devices named "Arduino 101" did not get recognized (yellow signs).

facchinm, 2016-03-21:
Hi @xinjincanada ,
the 101 core should install the drivers since it contains a `post_install.bat` file which is executed as the last step of the core installation (see https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/contributions/packages/ContributionInstaller.java#L145).
Can you try, launching the IDE with `arduino_debug.exe`, to uninstall and reinstall the 101 core, pasting here the console output?
The fact that you can't see the COM3 port in the IDE probably means that the port is not well configured. Can you also try to connect to another USB port and see if it works correctly?

xinjincanada, 2016-03-21:
Thank you. I tried uninstall 1.04 core and reinstall 1.05 core using arduino_debug, the console output is:

Loading configuration...
Initializing packages...
Preparing boards...
Starting...
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT

I also tried other USB ports, nothing changed, the find new hardware wizard could not install the driver properly. The IDE still only has COM1 and I had virtual port COM3.

By the way, where is post_install.bat file? May I run it manually?

facchinm, 2016-03-21:
You can find the file in `C:\Documents and Settings\user\Local Settings\Application Data\Arduino15\packages\Intel\tools\arduino101load\1.6.4+1.18`and you can start it manually; it should trigger a password request and install the correct drivers.
The file comes from here https://github.com/01org/intel-arduino-tools/blob/windows/post_install.bat

xinjincanada, 2016-03-21:
Thanks again! Just tried running the post_install.bat manually, and after running it, connecting to the USB port of the board, the find new hardware wizard still comes up, and still cannot find driver either automatically or manually specifying location:
C:\Documents and Settings\user\Local Settings\Application Data\Arduino15\packages\Intel\tools\arduino101load\1.6.4+1.18\drivers

PaulStoffregen, 2016-03-22:
Which version of Windows?

xinjincanada, 2016-03-22:
WIN XP SP3 (media center edition).

xinjincanada, 2016-03-24:
Will a bug fix come soon?

facchinm, 2016-03-25:
Probably not, XP is 14 years old and SP3 is 8 (at least).
I understand your concern but we can't track the same issue in different places, so I'm closing this one, let's keep the one on the forum alive (and let's try to find a workaround)

closed 2016-03-25 by facchinm

xinjincanada, 2016-03-25:
Understand. Did Intel provide a raw driver release that I can install manually? Or were there any old versions of 101 Core that was able to install a driver?