arduino/Arduino #4725: seperate mcu source from ide source.

Duality4Y, 2016-03-20:
is there any way to do this?
because it would make it a whole lot easier if you could just copy the libraries and all the arduino sources into  you project and distribute with that so you could always distribute a working version that compiles without having to download the ide.

Duality4Y, 2016-04-30:
I meen wouldn't it be nice to also have a seperate avr-core?
I don't think it would be that much work to do?
sure it comes pre-installed. but that can be solved with some script downloading it at compile time for the ide?

facchinm, 2016-07-19:
Hi @Duality4Y , the only reason `avr` and `sam` core are still in-tree is that taking them out would invalidate a lot of pending PR. If we had more time the core would be already standalone (which is what we are doing with the libraries and https://github.com/arduino-libraries repo)

per1234, 2017-07-04:
Closing as duplicate of https://github.com/arduino/Arduino/issues/5976 because, even though this one came first, that one has a bit more information.

closed 2017-07-04 by per1234