arduino/Arduino #4726: Mysterious RX pulses on UART connect on OS X Arduino Due

bcr, 2016-03-21:
Arduino IDE 1.6.8, Arduino Due, Mac OS 10.11.3

I am seeing eight mysterious pulses on the RX line when I connect to the serial port using multiple client libraries (Python, JavaScript as well as the built-in Serial Monitor in the IDE). About 78-79us apiece, sampled at 1MS/s with a Logic Pro 16.

[image]

These eight pulses when interpreted at 57600 baud will jam the Firmata firmware. And they happen on every connection.

This is using a new install of the Arduino 1.6.8 IDE and with multiple sketches (the normal "Blink" sketch will reproduce this also).

Repro steps on my machine:

1. Install any sketch
1. Start a logic analyzer if you want to catch it
1. Go to Serial Monitor. I have mine configured for 57600 baud, Newline line ending, but it doesn't matter
1. If you want, close and repeat step 3
1. Note pulses each time you connect to the serial port

Any suggestions for diagnosing this? It sounds like it's serial driver-level in some way.

facchinm, 2016-03-21:
Hi @bcr ,
are you using native or programming port? If you are on native, can you try to switch back to IDE 1.6.7 and see if the mysterious signal is still there?

cmaglie, 2016-03-21:
IIRC this has been fixed a long time ago in the mega16u2 firmware (the chip that does USB-2-serial on the programming port).

https://github.com/arduino/Arduino/commit/976f7588b7f68bd3c5fce52c15ccfe738589085a

It may be that you have a very old Due?

PaulStoffregen, 2016-03-21:
Or a Chinese clone?

bcr, 2016-03-21:
@PaulStoffregen Yeah, that's always a potential factor, isn't it -- I tend to be fairly careful about this, I got it from SparkFun or someone reputable when I got it, markings look legit.

@cmaglie This is likely the case -- my Due is from when it first came out, so if there's no automatically applied firmware update for the 16u2 then this is likely my problem.

I'll research how to update the firmware, the changelist above seems likely to fix the problem, I'll close it and reopen if it doesn't.

Thanks for taking the time to look at this.

closed 2016-03-21 by bcr

cmaglie, 2016-03-22:
@bcr 
maybe a workaround in the sketch is easier for your purposes compared to the tedious process to update the 16u2 firmware via DFU or ISP, I imagine changing the line:

```C++
Firmata.begin(57600);
```

with something like

```C++
Serial.begin(57600);
delay(1000);
whlie (Serial.read() != -1) { /* Eat garbage */ }
Firmata.begin(Serial);  // Serial1.begin(57600);
  // Firmata.begin(Serial1);

```

don't know, just a though...

bcr, 2016-03-22:
My plan was to submit a patch to Firmata to time out `START_SYSEX` (0xF0) so the protocol becomes responsive.

In general Firmata seems to be sensitive to the state and order that the firmware and client software is running -- your suggested fix is quick to implement on my side, which is good. But if the firmware is already running, the time to eat garbage is over when the client connects (generating garbage). The suggested fix helps a little, but if you start the firmware and then run the client within 1000ms the fix will eat potentially legit commands from the client. If you run the client after 1000ms, uneaten garbage and stuck in SYSEX.

Thanks for the suggestion -- I'll keep it in mind.