arduino/Arduino #4728: Board manager won't work on macbook pro, but it will work on iMac with the exact same operating system. I was trying to add ATtiny 44 to the hardware list. like I say, it worked on the iMac but refuses to work on the macbook pro.

Stevensea, 2016-03-21:


cmaglie, 2016-03-21:
Can you please add some more details?

- MacOSX version
- Arduino IDE version
- URL of the external core you're trying to install
- What do you mean with "won't work"? what do you expect and what is happening instead?
- There are some errors displayed that you can copy and paste here?

thanks.

Stevensea, 2016-03-26:
I solved the problem:-) My initial download must have been corrupted somehow. It took me some time to workout how to fully uninstall it from my OS X system. But when I did that and reinstalled it, the problem was gone. 

Regards Steven.

Breidenbach, 2016-03-26:
@Stevensea

Can you document the steps needed to uninstall completely?  I’d like to do that also, I am hopeful that it would solve my problems also.

Thank you,

Hal

sandeepmistry, 2016-04-04:
Closing this for now, until we have steps to reproduce.

closed 2016-04-04 by sandeepmistry