arduino/Arduino #4729: Why wont this for loop work

ninjacharlie0303, 2016-03-21:
**   for (int j=1 j<=10; j+j+1 ){
  
  digitalWrite(blueLEDpin,HIGH);  //turns blue LED on
  delay(blueWaitTimeOn);  //delays the code
  digitalWrite(blueLEDpin,LOW); //turns blue LED off
  delay(blueWaitTimeOff);  //delays the code 
  }**

blfuentes, 2016-03-21:
Check the section "The for loop" here
[http://www.cplusplus.com/doc/tutorial/control/](url)

-You are missing a semicolon between int j=1 and j<=10
-increase section j+j+1 is not "correct". It creates an infinite loop because j value is not being modified. In case you want to increase its value by 1 each time just use j++

```
for (int j=1; j<=10; j++ )
{
    digitalWrite(blueLEDpin,HIGH); //turns blue LED on
    delay(blueWaitTimeOn); //delays the code
    digitalWrite(blueLEDpin,LOW); //turns blue LED off
    delay(blueWaitTimeOff); //delays the code 
}
```

ninjacharlie0303, 2016-03-21:
i have tried this and it does not work. this is now my code:

 fo (int j=1;j<=10; j++ ){
  
  digitalWrite(blueLEDpin,HIGH);  //turns blue LED on
  delay(blueWaitTimeOn);  //delays the code
  digitalWrite(blueLEDpin,LOW); //turns blue LED off
  delay(blueWaitTimeOff);  //delays the code 
  }

ninjacharlie0303, 2016-03-21:
i am using the latest version  of arduino

blfuentes, 2016-03-21:
I suggest you to post the complete code somewhere like pastebin and show the wiring.
The syntax of the loop is correct. If the behavior is not the expected the problem is somewhere else.
-Values for blueLEDpin, blueWaitTimeOn ...

ninjacharlie0303, 2016-03-21:
The led are blinking but the second led has stopped as soon as i tried to put the for loop in

blfuentes, 2016-03-21:
I'm sorry but without the complete code and schema for your current wiring it is not possible to find the problem

ninjacharlie0303, 2016-03-21:
i have worked it out i added in something different and it worked it was the for missing the "r" thank you

ninjacharlie0303, 2016-03-21:
why wont the red LED blink?? this is my code for the whole thing

int blueLEDpin=11;
int blueWaitTimeOn=900;
int blueWaitTimeOff=100;
int redLEDPin=13;
int redWaitTimeOn=100;
int redWaitTimeOff=900;
int numBlueBlink=3; //number of times the blue LED blinks

void setup() {

  pinMode(blueLEDpin,OUTPUT);
  pinMode(redLEDPin,OUTPUT);
}

void loop() {

   for (int j=1;j<=numBlueBlink; j+1 ){
  
  digitalWrite(blueLEDpin,HIGH);  //turns blue LED on
  delay(blueWaitTimeOn);  //delays the code
  digitalWrite(blueLEDpin,LOW); //turns blue LED off
  delay(blueWaitTimeOff);  //delays the code 
  }

  digitalWrite(redLEDPin,HIGH); //turns red LED on
  delay(redWaitTimeOn);  //delays the code
  digitalWrite(redLEDPin,LOW);  //turns red LED off
  delay(redWaitTimeOff);  //delays the code

}

blfuentes, 2016-03-21:
The loop never ends. You wrote j+1. That is incorrect. The variable j is not being increased at all.
Try write j++.

ninjacharlie0303, 2016-03-21:
thank you the guy i'm watching said's to put that.

closed 2016-03-21 by sandeepmistry

cmaglie, 2016-03-21:
This issue tracker is for tracking bugs in Arduino code (IDE and cores) not for support to beginners.

The next time for this kind of questions please use the forum -> https://forum.arduino.cc

Thank you!