arduino/Arduino #4730: Request for help to resolve WINC1500 TLS bug stopping Arduino Wifi 101 from connecting

mashikawa, 2016-03-22:
Hi,  I purchased the Arduino Wifi 101 shield for an IoT project and found that I could not connect to certain servers via SSL.  Some worked (www.google.com:443) and others did not (plot.ly:443).

Using Wireshark to monitor the TLS handshake I found that the WINC1500 hardcodes the TLS Client Hello value to 1.2 ( since firmware version 9.3.0 ), although the device does not handle ciphers longer than 256bits ( SHA-256 ).   This actually makes it only TLS1.1 compatible.  Some servers now implement SHA-384 and SHA-512 ciphers as part of the TLS1.2 protocol and they can respond to the TLS1.2 request with these ciphers, which the WINC1500 (Wifi101) cannot decrypt and so the connection fails.  I have recreated the issue using OpenSSL and have attached the output to this request for info.  

I raised a support request with Atmel ( Case 00037229 ), suggesting they allow the TLS value to be modified by the user via the software and they responded with the following message:

_Created By: Anu Ramakrishnan (3/21/2016 4:09 AM)
[Recipients: Michael Kelsall]

Hi Micheal, 
Yes, we do not support SHA-384 and 512. Thank you for the feedback regarding the usage of TLS1.2 mode. We have raised a bug internally for this and we will follow up to get it fixed in future releases. 
We do not share firmware sources of the ATWINC1500, sorry for the inconvenience. 
Regards, 
Anu
_
**This is a good response but I am now worried the issue may sit on a development list with low priority for an extended period of time.....   Can anyone help to progress this as I am sure the problem will get worse as more website implement longer ciphers to improve security.
**
Regards,
Michael.

[OpenSSL_ATWINC1500_TLS_Examples.txt](https://github.com/arduino/Arduino/files/183290/OpenSSL_ATWINC1500_TLS_Examples.txt)

facchinm, 2016-03-22:
I think this should be moved to https://github.com/arduino-libraries/WiFi101 giving it top priority since it's going to become a very serious problem. Thanks for reporting

facchinm, 2016-03-22:
Moved to https://github.com/arduino-libraries/WiFi101/issues/48

closed 2016-03-22 by facchinm

cmaglie, 2016-03-22:
For the record: we have already reported this problem to Atmel more than once in the last two months. Unfortunately there isn't much we can do except waiting for a new release of the firmware.

mashikawa, 2016-03-22:
If I knew this board had such a fundamental issue, I would not have purchased it.   Based on your comment I should return it for a refund as it will not be resolved in any reasonable timescale.   This issue essentially turns the card back to a basic Wifi shield.

cmaglie, 2016-03-22:
>  Based on your comment I should return it for a refund as it will not be resolved in any reasonable timescale

I understand your concerns, my comment is just a genuine report of the current state for this issue. The latest information we have is that Atmel is working on that, but we haven't a fixed deadline for the solution.

mashikawa, 2016-03-30:
cmagile,  I received the following request from Atmel Support, so could you please provide the incident numbers you referred to, so we can try to escalate this issue:

"Please respond above this email, avoid inline comments

Commented by Anu Ramakrishnan (Atmel) 2016-03-28 23:56 PDT
[Recipients: Michael Kelsall]

Hi Michael, 

I am following up on this internally. We will get back to you as soon as we have further information. 
Meanwhile, can you provide the case numbers/mail thread of the issues raised by the Arduino team in this regard, for our reference? 

Regards 
Anu "

cmaglie, 2016-03-30:
It's nice that someone from the Atmel's helpdesk is trying to escalate the issue but, as a partner of Atmel, Arduino can talk directly with the second level support, so we have no case number to show... as far as I remember, we never used the frontline helpdesk.

cmaglie, 2016-03-30:
/cc @ThibaultRichard