arduino/Arduino #4735: Problem with DHTtester sketch

efidje, 2016-03-22:
I have just tested the DHTtester sketch in Arduino 1.6.5. It compiles ok if I use a Nano board but throws an error if I select a NodeMCU 1.0 board. 
[DHTtester_error.txt](https://github.com/arduino/Arduino/files/183608/DHTtester_error.txt)

facchinm, 2016-03-22:
Probably it's a problem of your setup since IDE 1.6.9 + ESP core v. 2.1.0 compiles correctly.
Try updating the IDE and the ESP core and it should solve automatically. 
For ESP-related problems please use https://github.com/esp8266/Arduino  bug tracker, thanks :wink:

closed 2016-03-22 by facchinm

efidje, 2016-03-24:
Hi,

Where do I get IDE version 1.6.9? On the Arduino site there is only 1.6.8...

Rgds
Endre

facchinm, 2016-03-24:
Hi Endre,
you can download a work-in-progress snapshot of 1.6.9 on Downloads page under "hourly build".