arduino/Arduino #4738: Compiler verbose output - no word / line wrap.

VaclavSal, 2016-03-22:
I have been living with the issue of compiler verbose output not line wrapping since I started using 1.6.5 , when it quit line wrapping , and up. I did submit a report on it and it got no reaction. Fine.

Now 1.6.8. changed the directory structure and it broke using modified print.h for printf function. 
Since the new "arduino_builder" is not documented ,maybe I missed it, I now have to muddle thru "extra wide" output  to find out how can I get back to correct "print.h" to modify. 

I would really appreciate if someone took pity on me and put the "word / line wrap" back .

steronydh, 2016-04-13:
+1 It's a pain having to scroll around the output window to find the useful part of an error message, which is invariably out of view.

Please can we have word wrap in the output window?

lmihalkovic, 2016-04-13:
#4411 lists alternative improvements that IMHO should be made to this compiling business... However you try to dummy ghk gs down, this app is still for developers, and cleanly dealing with errors should be top priority.
The problem will likely go partially away with Create because someone else wrote the code to do that, so all they have todo is use it.

per1234, 2022-09-22:
We are now tracking this at https://github.com/arduino/arduino-ide/issues/1486

closed 2022-09-22 by per1234