arduino/Arduino #4740: Missing printf link in 1.6.8.

VaclavSal, 2016-03-22:
I have decided to post this separately from my previous post, just to keep one subject per issue.

I have been using printf since 1.5.x after modifying core print.h.
Now using 1.6.8. and this simple test code turns the Serial.printf red indicating the there is valid printf function somewhere.

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.printf(__FILE__);
}

void loop() {
  // put your main code here, to run repeatedly:
}

Than I get this compiler error 

C:\Documents and Settings\Vaclav\My Documents\Arduino\sketch_mar21c\sketch_mar21c.ino: In function 'void setup()':

<command-line>:0:8: error: 'class UARTClass' has no member named 'iprintf'

C:\Documents and Settings\Vaclav\My Documents\Arduino\sketch_mar21c\sketch_mar21c.ino:4:10: note: in expansion of macro 'printf'

   Serial.printf(__FILE__);

          ^

exit status 1

Error compiling for board Arduino Due (Programming Port).

Any help would be appreciated, and PLEASE no "post it on Arduino.cc"  I am banned for life and it would be over their heads anyway.

matthijskooijman, 2016-03-22:
Serial / Print does not support printf. If you modified Print.h to support it, you're on your own, we cannot support you with modifications like this. My suggestion would be that you either did not re-apply the Print.h modification again, or perhaps applied it to the AVR core only, while you are now using the SAM core for the Due.

closed 2016-03-22 by matthijskooijman

VaclavSal, 2016-03-22:
Than you should communicate this to Arduino.cc since the printf modification is on their site. 
Actually it is not that, you do not know how to fix it and using the "it is not  supported" crutch excuse.
I have already figured it out.  
Still would like to know here this error  came from :
0:8: error: 'class UARTClass' has no member named 'iprintf', 
but I DO NOT NEED to know .