arduino/Arduino #4741: Seeed unlicensed fork of 1.6.0 iDE as modified version for Rephone

jamieisboss, 2016-03-22:
Not what you'd want from one of the Genuino manufacturing partners i Guess, breaking he GPL, removing the license and not releasing the changed code, also making it so the modified version only works under windows, then calling it open source and  Arduino-compatible in a kickstarter campaign then ignoring people pointing out the issue

- [Sorry, here it is](https://github.com/Seeed-Studio/Arduino_IDE_for_RePhone)
- [And the issue thats been open there since december asking for the code](https://github.com/Seeed-Studio/Arduino_IDE_for_RePhone/issues/1)

Apologies is this isnt the right place to put this as its not a bug or code issue but i wasn't sure where else to go, they probably need bringing up to speed with the board and library managers more than it being an attempt to be malicious. It just feels like the way people were doing things before these new releases, except removing the license file is pretty bad,

xiongyihui, 2016-03-25:
There are probably some misunderstood. 
[Seeed-Studio/Arduino_IDE_for_RePhone] contains release package files from arduino.cc. It's not a re-compiled version. The license file is not included in the release package, so it is not removed. Anyway, the license file should be added. Sorry for that.

jamieisboss, 2016-03-26:
Yeah, I messaged back and forth with them, re and they explained. 

Basically renaming the .exe and _not_ saying it _hadn't_ been recompiled was confusing, it's put a few people off as it looked like there was a recompiled version for windows and nothing else. This should  help Mac and Linux users understand how to use the boards through the IDE and get more people using a cool product, if they get the code packaged to just be added through the board and library managers then it'll be easy. 

Maybe a single Seeed board manager url for all Seeed boards and the Same for libraries would be easier for end users now there are more Seeed boards than ever before; so the end user would only have one url to add then they could just install what they need but that'd only be possible when the packaging was all done in the right format, so couldn't be done yet if the rephone was to be included.

To be honest, that was one of the most straightforward, nice experiences i've had when i've pointed out something like this, which is nearly always a misunderstanding, everyone makes mistakes occasionally, it's how you respond that counts and the rephone team saw the issue and did everything they could straight away, which was the best thing they could have done,. Also, thank you for the license additon, not that it's my project, but making sure the licences are used properly protects everyone in the long term so I try and help by pointing out when they aren't. 

Proved in a new way to me that Seeed are a great company.

xiongyihui, 2016-03-31:
Just add experiential rephone sdk for board manager - https://github.com/Seeed-Studio/Seeeduino-Boards/raw/master/package_seeed_rephone_index.json

Tested with Arduino IDE 1.6.4, while v1.6.7 doesn't work

jamieisboss, 2016-03-31:
Ok thanks!

closed 2016-03-31 by jamieisboss