arduino/Arduino #4742: Error compiling arduino code using 1.6.8 under linux on Athlon XP2800, same as issue #4251

mrchadwickusa, 2016-03-22:
I installed 1.6.8 for linux under Puppy Linux Slacko 5.7, on an Athlon XP2800 processor.  I have the same instruction error happening as in issue 4251, which was supposedly resolved in the latest releases. Perhaps the information didn't make it to those responsible for compiling the linux versions?

Arduino: 1.6.8 (Linux), Board: "Arduino Pro or Pro Mini, ATmega328 (3.3V, 8 MHz)"

SIGILL: illegal instruction
PC=0x81118c9

goroutine 1 [running, locked to thread]:
math.init·1()
	/usr/local/go/src/math/pow10.go:34 +0x19 fp=0x18435f84 sp=0x18435f80
math.init()
	/usr/local/go/src/math/unsafe.go:21 +0x3f fp=0x18435f88 sp=0x18435f84
strconv.init()
	/usr/local/go/src/strconv/quote.go:455 +0x45 fp=0x18435fa4 sp=0x18435f88
io/ioutil.init()
	/usr/local/go/src/io/ioutil/tempfile.go:95 +0x4a fp=0x18435fc4 sp=0x18435fa4
github.com/go-errors/errors.init()
	/home/jenkins/jenkins/jobs/arduino-builder-linux32/workspace/src/github.com/go-errors/errors/stackframe.go:102 +0x3f fp=0x18435fc8 sp=0x18435fc4
main.init()
	/home/jenkins/jenkins/jobs/arduino-builder-linux32/workspace/main.go:395 +0x3f fp=0x18435fcc sp=0x18435fc8
runtime.main()
	/usr/local/go/src/runtime/proc.go:58 +0xbd fp=0x18435ff0 sp=0x18435fcc
runtime.goexit()
	/usr/local/go/src/runtime/asm_386.s:2287 +0x1 fp=0x18435ff4 sp=0x18435ff0

goroutine 2 [runnable]:
runtime.forcegchelper()
	/usr/local/go/src/runtime/proc.go:90
runtime.goexit()
	/usr/local/go/src/runtime/asm_386.s:2287 +0x1

goroutine 3 [runnable]:
runtime.bgsweep()
	/usr/local/go/src/runtime/mgc0.go:82
runtime.goexit()
	/usr/local/go/src/runtime/asm_386.s:2287 +0x1

eax     0x23
ebx     0x826b3d8
ecx     0x18400140
edx     0x826b3d8
edi     0x18435f80
esi     0x18435f8c
ebp     0x1843c03d
esp     0x18435f80
eip     0x81118c9
eflags  0x10202
cs      0x73
fs      0x0
gs      0x3f
arduino-builder returned 2

Error compiling for board Arduino Pro or Pro Mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thanks for attending to this! The solution was to compile with GO386=387 option enabled to disallow the use of SSE2 instructions.

per1234, 2017-07-04:
Per https://github.com/arduino/Arduino/issues/4251#issuecomment-199909541, the SSE2 fix was only applied to the Windows build of Arduino IDE 1.6.8.

I suspect this issue has been fixed since the time of this report. @mrchadwickusa do you still have this problem with the latest version of the Arduino IDE?

per1234, 2017-08-22:
Closing due to lack of response from @mrchadwickusa.

closed 2017-08-22 by per1234