arduino/Arduino #4746: Default Gateway not working in Routed Network with Static IP's unless myDns used - DHCP works fine.

andysoondyba, 2016-03-23:
Issue when trying to connect to devices behind different networks.
This is not a home NAT'd setup. Two or more different networks connected with routers and default gateways.

Static IP settings don't establish a route/arp table when checking with Wireshark.
If the exact same settings however are received from DHCP, all is OK.

The addition of myDns solves the issue. It doesn't matter what I set the DNS IP address to, it just needs it. My project however doesn't require any DNS services.
After adding this is can see the route/arp requests exchange in Wireshark and the connection establishes.

What is myDns doing in the back ground to correct this action? From a comment on this topic in the forums and bug # 4616 it might be related to extra info that DHCP obtains that's not always being set when statically defining network setting.

When using DHCP and not declaring myDns the sketch will still work.
I'm assuming the data is still being collected in the library's inner workings.

facchinm, 2016-03-23:
Hi @andysoondyba,
- Which shield are you using?
- IDE and Library version?

andysoondyba, 2016-03-23:
Hello,
Freetronics ethermega with w5100.
Arduino 1.6.7Ethernet version 1.1.1(read from library.properties)

Sent from my Samsung Galaxy smartphone.<div>
</div><div>
</div><div>-------- Original message --------</div><div>From: Martino Facchin <notifications@github.com> </div><div>Date: 23/03/2016  4:55 pm  (GMT+08:00) </div><div>To: arduino/Arduino <Arduino@noreply.github.com> </div><div>Cc: andysoondyba <andysoondyba@hotmail.com> </div><div>Subject: Re: [Arduino] Default Gateway not working in Routed Network with Static IP's unless myDns used - DHCP works fine. (#4746) </div><div>
</div>
Hi @andysoondyba,
- Which shield are you using?
- IDE and Library version?

---
https://github.com/arduino/Arduino/issues/4746#issuecomment-200254416

per1234, 2021-03-28:
Transferred to https://github.com/arduino-libraries/Ethernet/issues/156

closed 2021-03-28 by per1234