arduino/Arduino #4747: Two Estonian languages

hasso, 2016-03-23:
There is two Estonian languages in Arduino - et and et_EE. It doesn't make any sense (there is no Estonian variants spoken in other countries), are just copies and the situation is just confusing for users.

The right thing to do is just remove et_EE from Transifex and from code.

hasso, 2018-05-18:
Can I get a little bit attention with this, please? :)

facchinm, 2018-05-21:
Hi @hasso ,
removing et_EE from the code is trivial (removing this line should be enought https://github.com/arduino/Arduino/blob/master/app/src/cc/arduino/i18n/Languages.java#L66 ). About transiflex we probably need some help from @cmaglie :slightly_smiling_face:

cmaglie, 2018-05-21:
You beat me on this one, I've just started looking on transifex...

closed 2018-05-21 by cmaglie

buhtz, 2023-09-06:
Dear @hasso ,
my apologize for contacting you this way but I wasn't able to find a email address or something else.

I assume you are a native speaker of Estonian, right?

I'm looking for a translation of a [100 words text](https://translate.codeberg.org/translate/backintime/common/et/?q=hello) into Estonian in the context of a backup software named [Back In Time](https://github.com/bit-team/backintime). The text is addressed to the users and tries to attract them contributing to the translation of the software they use them self.

Can you help us out here?

commit b70e876c (2018-05-21): Removed et_EE localization.

Fix #4747