arduino/Arduino #4748: Who is on first ? AKA why two copies of Arduino15 directory in Windows

VaclavSal, 2016-03-23:
Would it be possible to get some info why there are two sub-directories "Arduin15" in  Windows?
They do not contain same info after "Board manager" is used. 

Also  - and I believe , right or wrong,  that this is related to the above.
If you backpedal  to older versions  of IDE from 168,  what version of "board package" is safe to select? 
For example in IDE version 168 Due board versions are up to 1.6.7 , but using 1.6.7. won't compile when going back to IDE 164.  
.

pfeerick, 2016-03-30:
Do you mean the two different versions of Arduino15 (on Windows 10, one at %userprofile%\AppData\Local\Arduino15 and the other at %userprofile%\AppData\Roaming\Arduino15). If so, I believe that the Roaming folder was used until  Arduino 1.6.5, after which they switched to the Local folder after much debate on the developer mailing list (mainly due to the fact that the Roaming folder is network/internet synced, whereas Local stays local, IIRC). So post-1.6.5, you should be able to remove the Arduino15 folder in Roaming without any issues if you so desire. 

I can't help you with when it is safe to have a board/core newer in version than the IDE - I would suspect the general rule of thumb would it it isn't safe, as newer cores would be relying on changes to the IDE, but it would really be a case of trial and error.

VaclavSal, 2016-03-30:
Thanks, it definitely WAS a trial and many errors when I foolishly went from perfectly working 1.6.6 to 1.6.8. 
Now I am back to 1.6.4 and it does the job for me. 

After seeing few "problems" people have with 1.6.8. I am beginning to wonder how Arduion.cc whose unwritten business model is to cater to beginners keeps publishing (untested) stuff like this?
For me it would be a great turn-off if I, as a beginner, start with 1.6.8. and run into so many issues. 
This is not the first time "upgraded IDE" was buggy, it looks as developers and Arduino.cc are not connected as business partners.

closed 2016-03-30 by VaclavSal