arduino/Arduino #4751: template <typename Generic> can not compile with the lastedt version 1.6.8.

EagleSmith, 2016-03-24:
template <typename Generic>
void DEBUG_WM(Generic text) {
  if (_debug) {
    Serial.print("*WM: ");
    Serial.println(text);
  }
}

it's working fine with Previous version(1.6.7)

facchinm, 2016-03-25:
Hi @EagleSmith ,
I've tested IDE 1.6.7 with your example and it fails to compile, too. Can you give more hints? Like target board, verbose compilation output etc etc

cmaglie, 2016-03-25:
just a blind guess, but could be that the function prototype spans over two lines?
What happens if you write:

```C++
template <typename Generic> void DEBUG_WM(Generic text) {
  if (_debug) {
    Serial.print("*WM: ");
    Serial.println(text);
  }
}
```

?

nickgammon, 2016-06-04:
I was about to report something similar when I found this report. If you remove the line break it compiles OK. With the break, the preprocessed code is:

```c++
template <typename Generic> 
#line 4 "/tmp/arduino_modified_sketch_455371/sketch_jun04c.ino"
void DEBUG_WM(Generic text) ;
#line 11 "/tmp/arduino_modified_sketch_455371/sketch_jun04c.ino"
void setup();
#line 16 "/tmp/arduino_modified_sketch_455371/sketch_jun04c.ino"
void loop();
#line 4
void DEBUG_WM(Generic text) {
  if (_debug) {
    Serial.print("*WM: ");
    Serial.println(text);
  }
}
```

Arduino .ino file was:

```c++
bool _debug = true;

template <typename Generic> 
void DEBUG_WM(Generic text) {
  if (_debug) {
    Serial.print("*WM: ");
    Serial.println(text);
  }
}

void setup() { }
void loop() { }
```

Removing the #line directives the problem is it was expanded out to:

```c++
template <typename Generic> 
void DEBUG_WM(Generic text) ;
void setup();
void loop();
void DEBUG_WM(Generic text) {
```

The template line is not repeated. The prototype is OK, but the "real" function (definition) does not get the template line. The regexp or whatever-you-use to find a function declaration needs to allow for a line break in the declaration.

---

IDE 1.6.8 and 1.6.9.

VioletGiraffe, 2017-02-23:
I confirm this is still present in 1.8.1.

facchinm, 2017-02-23:
Should be solved by https://github.com/arduino/arduino-builder/pull/182, @VioletGiraffe could you test it? Thx

VioletGiraffe, 2017-02-23:
@facchinm : how can I test it? Sorry, I'm new around here :)
Apparently, I need to install Go and build the builder. Can I then use this new builder by replacing the "C:\Program Files (x86)\Arduino\arduino-builder.exe" with this new one?

facchinm, 2017-02-23:
No problem :slightly_smiling_face: You don't need to install go, simply download the build produced by @ArduinoBot (http://downloads.arduino.cc/PR/arduino-builder/arduino-builder-182.zip) and replace the `arduino-builder` executable in the IDE installation folder with the correct one for your platform.

VioletGiraffe, 2017-02-23:
Good to know! Yes, that indeed does solve the problem.

per1234, 2017-07-01:
I also confirm the issue is solved by https://github.com/arduino/arduino-builder/pull/182 so I'm going to close this. @EagleSmith if you test the https://github.com/arduino/arduino-builder/pull/182 build and find that the problem still occurs please post a minimal complete sketch that demonstrates the problem in addition to the information requested at https://github.com/arduino/Arduino/issues/4751#issuecomment-201326219 and I'll reopen.

closed 2017-07-01 by per1234