arduino/Arduino #4753: Problem acces OK button on preference window

yvescordier, 2016-03-24:
with the ide (actualy i use the 1.6.8 version) , the preference window  has no scrollbar (vertical). So it is impossible to acces to OK button on the bottom of the window on a small PC (old recycle for courses) with lower resolution on y (600x1024). Wou
[image]
ld it be possible to reorganize the items of this window to be able to acces then on 10'' old notebook. Ther is a lot of room unused between lines checkboxes  and ...
another way could be to put the OK and disable button on the top of the box.

lmihalkovic, 2016-03-24:
@yvescordier this was my own thinking on the preferences: #4421, #4417, [#4102/comment](https://github.com/arduino/Arduino/pull/4102#issuecomment-171289157): the single page has been split into more logical smaller pages accessible via the left hand-side list. I think the arduino team's original argument that 'users are so non-tech that they can't be overwhelmed with too many choices' has over the years turned into an efficient justification for doing the bare minimum, especially considering how the IDE (not the libraries) could be construed as having entered _de-facto sunset mode_.

yvescordier, 2016-03-26:
Sorry but english is not my mother language. My problem is not to split the informations and checkboxes in two or three pages but the problem is to be able to access all informations of the preference window on a screen with low resolution (1024*600). On such screen, it is not possible to acces the OK button to validate the choices. It is why I propose to have a slimerst box of about 500 point heigh (this could be done by elimating usless zones).

regards
yves cordier

lmihalkovic, 2016-03-26:
```
this could be done by elimating usless zones
```
... or by avoiding to cram together things that have nothing in common, but result in a window too big for everything to be visible on a smaller resolution monitor. But the arduino designers likely has a plan for what they want to do.
Cheers.

Reminds me of this architect friend who used to say nobody could pay him enough to make him design a toilet in the center of a living room, without a very serious reason...

facchinm, 2017-08-01:
Fixed by #6459

closed 2017-08-01 by facchinm

yvescordier, 2017-08-13:
> Message du 01/08/17 11:48
> A : "arduino/Arduino"
> Copie à : "yvescordier" , "Mention"

sorry to have been late to try the modification but i Had'nt my computer avilable. 

due to function bar of  XP on the bottom of the screen, it sill be difficult to press the buttons

only 10 pixels of the button are visible

50 pixels less should be apeciate

is it possible to remove 2 or 3 trailing lines more at the bottom of the preference windows

 

Thank's for all

regards

yves cordier

facchinm, 2017-08-14:
Due to the padding of all the other elements it's truly difficult to get back other 50 pixels (there is always the "dual tab" chance, as I mentioned in #6459).
The main problem I think it's gone is that it was simply impossible to press the buttons. Now it's difficult but feasible (temporarily hiding the taskbar should help in the "desperate" cases of 1024x576 screens).

rin67630, 2018-11-27:
Temporarily hiding the taskbar does not help enought to get the button on screen.
It would have been so easy to put the 7 tick options side by side on two columns...

cjdg, 2019-05-29:
Its happening again on 1.8.9