arduino/Arduino #4754: CurieIMU documentation syntax uses wrong case

PaulStoffregen, 2016-03-24:
The many CurieIMU documentation pages have Syntax sections with "CurieImu".  It should be "CurieIMU".

For example:

http://www.arduino.cc/en/Reference/CurieIMUgetAccelerometerRate

    Syntax
    
    CurieImu.getAccelerometerRate()

SimonePDA, 2016-03-24:
Ok,  noted.  Leftovers from the former version.  I'll fix the reference asap.

SimonePDA, 2016-03-28:
All the occurrences have been fixed. Now the syntax is "CurieIMU.function()"

per1234, 2017-05-03:
@SimonePDA @agdl  almost every reference page for the CurieIMU library still has this problem.
https://www.arduino.cc/en/Reference/CurieIMU

The only ones that **DON'T** have the problem are:
- https://www.arduino.cc/en/Reference/CurieIMUdetachInterrupt
- https://www.arduino.cc/en/Reference/CurieIMUattachInterrupt
- https://www.arduino.cc/en/Reference/CurieIMUshockDetected
- https://www.arduino.cc/en/Reference/CurieIMUreadTemperature

Also, two of the links at https://www.arduino.cc/en/Reference/CurieIMU have typos:

https://www.arduino.cc/en/Reference/CurieIMUsgetDetectionThreshold
should be changed to:
https://www.arduino.cc/en/Reference/CurieIMUgetDetectionThreshold

https://www.arduino.cc/en/Reference/CurieIMUsgetStepDetectionMode
should be changed to:
https://www.arduino.cc/en/Reference/CurieIMUgetStepDetectionMode

akash73, 2017-07-17:
fixed thanks

per1234, 2017-07-17:
Thanks @akash73! There are still three pages with the "CurieImu" problem:

- https://www.arduino.cc/en/Reference/CurieIMUnoGyroOffset
- https://www.arduino.cc/en/Reference/CurieIMUgetGyroOffset
- https://www.arduino.cc/en/Tutorial/Genuino101CurieIMUOrientationVisualiser

SimonePDA, 2017-07-17:
Two pages weren't pushed properly, done. THe tutorial had an occurrence of CurieImu in Hardwared Required and was fixed as well.

SimonePDA, 2017-07-17:
I keep pushing, but it doesn't update. Investigating

SimonePDA, 2017-07-17:
After three attempts each, the pages finally got updated

closed 2017-07-17 by SimonePDA