arduino/Arduino #4755: Symbol inconsistanties for I2C between ARM, AVR cores...

WestfW, 2016-03-25:
The AVR Arduino variants define symbols SCL and SDA for access to the Arduino Pin numbers associated with those functions:
```
static const uint8_t SDA = 18;
static const uint8_t SCL = 19;
```
The ARM Arduino variants (Due AND zero) define symbols PIN_WIRE_SDA and PIN_WIRE_SCL instead:
```
#define PIN_WIRE_SDA         (20u)
#define PIN_WIRE_SCL         (21u)
```
This makes it difficult to write libraries or sketches that will compile for both types of platforms.
Probably, both platforms should define both sets of symbols, but at least the AVR platforms should get the new-style (and better) PIN_xxx #defines.

(In fact, there are a lot of nice PIN_xxx defines that show up in the ARM platforms/variants that should probably be back-ported to the AVRs...)

sandeepmistry, 2016-04-04:
This can be closed once #4814, #4813, and https://github.com/arduino/ArduinoCore-samd/issues/130 have been merged.

@WestfW let me know if I've missed anything.

agdl, 2016-07-07:
I'm closing this thank you!

closed 2016-07-07 by agdl