arduino/Arduino #4761: updated software and now getting error messages on code using

CamilleDorsa, 2016-03-25:
Hello,

I was using some MakeBlock test code to check out readings from a new ultrasonic sensor.  I updated the Arduino software when prompted.  Now when I run the same software, I get a lot of code errors and I cannot connect to the Arduino board on the robot. (this was standard library code that was working prior to the update) I have not had to connect to a specific board at this point, but I went to try that to see if it fixed the problem.  But the Arduino board I have, Makeblock Orion, is not in the list.  I have rebooted my computer a couple of times, now the code will compile and let me load the to robot.  What is going on?

VaclavSal, 2016-03-26:
**What is going on?**
See A.Cooper "inmates are running the asylum"

per1234, 2017-07-04:
Without more details it is impossible to investigate this issue so I'm closing it.

@CamilleDorsa, if you're still having this problem with the latest version of the Arduino IDE then please post the full output from the console window and I'll reopen this.

closed 2017-07-04 by per1234