arduino/Arduino #4762: File > Examples doesn't follow architecture matching library include priorities

per1234, 2016-03-26:
Using Arduino IDE 1.6.9 2016/03/25 02:33 with Windows 7

If there is an IDE bundled library with an `architectures` value in library.properties that matches the selected board's architecture and another library of the same name in the sketchbook that doesn't match the architecture then the IDE bundled library has include priority but the sketchbook library's examples are shown in **File > Examples**.

Here's a more complex real life example of this issue causing an error.
- **Tools > Board > Zero**
- **Sketch > Manage Libraries... > USBHost > Update** - this will install the SAM version of USBHost to {sketchbook}/libraries.
- **File > Examples > USBHost > ADKTerminalTest** -  {sketchbook}/libraries/USBHost/examples/ADKTerminalTest/ADKTerminalTest.ino is opened but the SAMD core version of the USBHost library will have include priority because of https://github.com/arduino/arduino-builder/commit/b2f7c72c05e2a1160866c0da8bca9fa9c896ce27.
- Attempting to compile this sketch for Zero fails.

facchinm, 2016-04-05:
Hi @per1234 
nice finding, the frontend problem is exposing a very bad behaviour (USBHost APis are not compatible between Due and Zero). Anyway, the issues should also be solved in the frontend side. 
I'll try to see what can be done :smile:

sandeepmistry, 2016-07-19:
Closed via #4828.

closed 2016-07-19 by sandeepmistry