arduino/Arduino #4763: 2 issues in arduino 's subdomains

Chichirovo, 2016-03-26:
Hello 

My name is Sami Drif

I'm web application security tester and i have found vulnerability (BUG)
in your website and it can cause damage your website

The first bug :

When t register an account , then i try to activate it and error message show

"Fatal error: Allowed memory size of 134217728 bytes exhausted (tried to allocate 3676663545 bytes) in /var/www/store/htdocs/ww/includes/functions/html_output.php on line 62"

This means it is Full path discloure 

the second bug :

url: https://store.arduino.cc/

This site uses a commonly-shared 1024-bit Diffie-Hellman group, and might be in range of being broken by a nation-state. It might be a good idea to generate a unique, 
2048-bit group for the site

Poc : Test with https://weakdh.org/sysadmin.html

I'm looking for your reply 

Thank you

mastrolinux, 2016-04-04:
Hi @Chichirovo thanks for reporting. We will solve the https cert tomorrow. The other issues requires a bit more time. Luca

mastrolinux, 2016-04-05:
Diffie Hellman issue has been fixed. Working on the other one right now. Will be fixed in 1-2 days.

facchinm, 2016-07-19:
@mastrolinux , can we close this?

Chichirovo, 2016-07-19:
Yes

facchinm, 2016-07-19:
Thanks @Chichirovo !

closed 2016-07-19 by facchinm