arduino/Arduino #4768: Color Mixing Lamp -

M2pz, 2016-03-27:
Greetings,

I found on the Color Mixing Lamp project page 59, that to avoid the following error, "Colour_Mixing_Lamp.ino:38:3: error: 'redValue' was not declared in this scope", a declaration of 
"int redValue = redSensorValue/4;" solves the issue in the coding. Thought I'd share, I'm loving the experience, thank you Arduino team.

agdl, 2016-04-12:
@M2pz I'm sorry but both in the book and in the online code I can see the declaration you state is missing. Can you please be more specific about this error?

M2pz, 2016-04-12:
@agdl, I was referring to the line 31 in page 59. I found that in my code, when I didn't declare that "redValue = redSensorValue/4" as an integer I got the error mentioned in my previous thread. But if I wrote the code as " int redValue = redSensorValue/4", it solves the issue. I hope this clarifies my message. Thank you.

per1234, 2017-07-04:
I don't have the book to look at but the p04_ColorMixingLamp example sketch does indeed have the declaration of `redSensorValue` and compiles fine:
https://github.com/arduino/Arduino/blob/1.8.3/build/shared/examples/10.StarterKit_BasicKit/p04_ColorMixingLamp/p04_ColorMixingLamp.ino#L37

closed 2020-05-23 by per1234