arduino/Arduino #4772: when I am using the compile guide, there are some errors as follows.

oneway3124, 2016-03-28:
The guide is from the site :https://github.com/arduino/Arduino/wiki/Building-Arduino
The erros is as below:
[image]

cmaglie, 2016-03-29:
The `JAVA_HOME` environment variable should point to a JDK but it seems that you set it to a JRE.

oneway3124, 2016-03-30:
[image]
can you tell me which linux version did you use? I am using ubuntu 14

cousteaulecommandant, 2016-05-06:
openjdk-8 is not available in the Ubuntu 14.04 ("LTS") repositories, only in 14.10 or newer (this has always been an issue in Ubuntu apparently).  You will need to install it separately, for example adding a PPA.  http://ubuntuhandbook.org/index.php/2015/01/install-openjdk-8-ubuntu-14-04-12-04-lts/ contains instructions on how to do this.

closed 2017-07-04 by per1234