arduino/Arduino #4774: Duplicate category Robotic & Robotics

rlogiacco, 2016-03-28:
Apparently there has been a typo somewhere ending up in two distinct categories, one called `Robotic` containing solely library `Svante`, one called `Robotics` containing two libraries, `IBot` and `WebServer`.

Please merge into one single category.

per1234, 2016-03-28:
All three of those libraries have had their category changed to a valid value already but the old releases still have the invalid values. IBot and WebServer are actually the same library but the first IBot release had the wrong name value. I wonder if it would be best to request that the library owner remove that release?

cmaglie, 2016-04-04:
(Sort of) duplicate of #4802, please continue the discussion there.

closed 2016-04-04 by cmaglie