arduino/Arduino #4775: Duplicate category Sensor & Sensors

rlogiacco, 2016-03-28:
For some reason, probably a typo, there are two distinct categories, one called `Sensor` containing only two libraries (`DallasTemperature` and ` SparkFun ZX Distance and Gesture Sensor`) and another one called `Sensors` containing numerous libraries.

Please correct the category name typo and merge the libs under one category.

per1234, 2016-03-28:
Which version of the Arduino IDE are you using? The issue is caused by older releases of those libraries that had the wrong category value. The value has since been corrected in both libraries. Using the hourly build this causes the versions with the invalid category value to appear under the **Uncategorized** category for me.

rlogiacco, 2016-03-28:
I'm not using the Arduino IDE, I'm instead parsing the libraries published JSON file

per1234, 2016-03-28:
The Arduino IDE Library Manager changes any invalid category value to `Uncategorized`. Valid category values(from https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) are:
>"Display", "Communication", "Signal Input/Output", "Sensors", "Device Control", "Timing", "Data Storage", "Data Processing", "Other"

So maybe you could employ the same behavior in your parsing?

rlogiacco, 2016-03-28:
I can and I will, but my advice is to fix the provided JSON: that will work for everyone and will fix the problem at its root.

per1234, 2016-03-29:
The entries in library_index.json are automatically generated from the library.properties file in each individual library repository every time there is a new version tag created so it isn't a matter of fixing the JSON file. To fix these issues either the library authors need to use valid category values or the code that generates the JSON would need to be changed to set invalid category values to `Uncategorized`. For more information see https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

rlogiacco, 2016-03-29:
So I believe the code that collects the `library.properties` to generate the `library_index.json` should verify the correct categorization and set the library to `Uncategorized` if the supplied value doesn't match any of the official categories. 
This way the _verification_ is performed from a central and unique point, and whenever you desire to add a category you don't need to force an update on the Arduino IDE to permit a new category.

Also, by looking at the logs, you will be able to communicate to the library provider if their library supplied properties are incorrect and decide if they should select a correct value or provide a fallback category.

IMHO a centralized can bring quite some value, first among all the categorization rules centralization, which is now delegated to the JSON consumers.

cmaglie, 2016-04-04:
Sort of duplicate of #4802, let's continue the discussion there.

closed 2016-04-04 by cmaglie