arduino/Arduino #4777: IDE 1.6.8 USB Problems - Affecting all 328 boards, Mega2560 and Due

TMRh20, 2016-03-29:
I'm using Windows 7 and after a bit of frustration and hardware testing, I've realized that I don't have multiple broken devices, IDE 1.6.8 just does not work for me, while all previous versions do. I've tried both the standalone and installed versions. 1.6.7 and earlier work fine.

I tested on a Nanov3.0, multiple Pro Minis w/FTDI cable, Duemilanove, and Mega2560 and Due clone.

I couldn't find an issue to match, but It seems this issue may be similar or related to https://github.com/arduino/Arduino/issues/4700

**Results of testing on multiple devices:**

**Due Clone:**
I can upload to the Due using the native USB port, but not the programming port.

```
Using library Servo at version 1.1.1 in folder: ...\Arduino\libraries\Servo 

Sketch uses 23,640 bytes (4%) of program storage space. Maximum is 524,288 bytes.
Forcing reset using 1200bps open/close on port COM12

Then it goes on to:

Verify 25908 bytes of flash
[                              ] 0% (0/102 pages)writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0x80000)
readWord(addr=0x400e0a08)=0x1
...
...

[=======================       ] 78% (80/102 pages)writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0x85000)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0x85100)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0x85200)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x20002000)
writeWord(addr=0x2000102c,value=0x85300)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x20001024,value=0x20001001)
go(addr=0x20001020)
read(addr=0x20002000,size=0x100)

SAM-BA operation failed
```

**Arduino Mega 2560:**

On the Mega, problems only arise with fairly large sketches, but it is consistent. The blink example seems to work fine.
```
avrdude: reading on-chip flash data:

Reading | #############avrdude: stk500v2_recv(): checksum error
##################################### | 100% 3.19s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x1912
         0xea != 0x28
avrdude: verification error; content mismatch

avrdude done.  Thank you.

```

**Arduino Nano v3.0 , Duemilanove & Pro Minis**

Basically they just sit there and resets until I open the Serial Monitor. The resetting starts as soon as I open the IDE v1.6.8, and interferes with uploads quite often, since it starts again as soon as the upload button is clicked. Besides the following, I was also getting an error, something about opening programmer with the pro minis, but can't replicate now.

```
Arduino Pro Mini w/FTDI cable:
         Using Port                    : COM13
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
avrdude: ser_open(): can't open device "\\.\COM13": Access is denied.

avrdude done.  Thank you.
```

It seems like something in the IDE is constantly trying to access or reinitialize the serial ports. I've tried reinstalling the IDE, FTDI drivers, rebooting, etc etc. and as soon as I closed the IDE, my Arduinos all stopped resetting. Opening V1.6.7 = no problems. I can provide more info if required.

pfeerick, 2016-03-30:
I've noticed the same thing, and for a few minutes there I thought I'd managed to break two of my Sparkfun RedBoards... as they kept resetting and blinking D13 every second or so, and was generally having issues after opening the serial monitor. Thankfully the reset issue went away after pluging the board into a external usb power bank, so I can say I am also a victim of this bug. Windows 10 Pro x64.

facchinm, 2016-03-30:
Ok, working on the fix (thanks to @TMRh20 and @ozligia on #4700 for the very exhaustive description)

facchinm, 2016-03-30:
@TMRh20 does the Due clone use an FTDI chip? The Mega2560 should mount an atmega16u2 if genuine...
Also, could you please test [this](https://github.com/arduino/Arduino/pull/4482#issuecomment-201348651) build? It stops the port scanning while uploading, maybe if it works (at least with the mega) we can try to spot the issue

TMRh20, 2016-03-30:
Yes, the Due is using an FTDI connector w/16u2 for the programming port, while the Mega has an 8u2. 

The Due and Mega test OK with that build, with very large sketches verifying consistently.

The resetting still occurs for the 328s, but now at a period of 1 LED blink/second. Normal activity resumes when the IDE is closed:

**1.6.8:**  LED flashed 3 times/second on startup & continues
**arduino-PR-4482-BUILD-531:** LED flashes 3 times quick, then 1 per second

As a side note, the very large sketches on Due and Mega actually seemed to be getting corrupted with 1.6.8 too. I can't prove it out, but I was getting some really strange behavior when verification failed.

pfeerick, 2016-03-31:
**arduino-PR-4482-BUILD-531** on a Sparkfun RedBoard (FTDI chip), with Windows 10: flashes the LED 3 times per second constantly unless the serial monitor is open, or uploading, but seems to stop the intermittent "Access is denied" error from avrdude: ser_open()

facchinm, 2016-04-07:
Solve by #4792

closed 2016-04-07 by facchinm