arduino/Arduino #4778: Unexpected Ctrl-Delete behavior in Arduino IDE

dtwaddell, 2016-03-29:
When I press Ctrl-Delete, I expect that the word in front of the cursor would be deleted, but instead, the whole rest of the line is deleted.

avargas-nearsoft, 2016-04-01:
It seems it's a problem from http://bobbylight.github.io/RSyntaxTextArea/ , the JTextArea implementation that Arduino IDE used in his Editor. I don't know if change it would be feasible, but I opine the same that you: it's a little confusing this feature. 
I will work in that and I will send a pull request.
What do you think about @ivanebernal @carlosMiguelCuevas

CarlosMiguelCuevas, 2016-04-01:
@dtwaddell , Why do you expect such behaviour? Where is that feature described? is it in the doc? I tried it and it seems that the behaviour of the Ctrl+Delete is to delete everything that is ahead of the cursor.

ivanebernal, 2016-04-01:
@CarlosMiguelCuevas I think It's because this behavior is present almost everywhere, except some text editors like notepad.

avargas-nearsoft, 2016-04-01:
Yes. It's a common feature for every text editor but I don't know why (neither where) RSyntaxTextArea is using this implementation.

dtwaddell, 2016-04-01:
@avargas-nearsoft  Right, notepad is the only text editor I know of that doesn't just delete the single word in front of the cursor.  But, Ctrl+Backspace doesn't work in notepad either.  Aside from that, this is the general behavior for Windows and other OSs.  Ctrl+Backspace deletes the word behind the cursor, Ctrl+Delete deletes the word in front of the cursor, and Ctrl+arrows moves left or right one word at a time.  

I checked a few programs to check, and the single word Ctrl+Delete behavior occurs in Microsoft Word, Google Chrome, Adobe Illustrator, Windows Explorer, Notepad++, and Processing (although they also don't do it perfectly either -- the extra space is still left behind, so pressing Ctrl+delete twice will still only get rid of one word + space).

avargas-nearsoft, 2016-04-01:
I found this class https://github.com/bobbylight/RSyntaxTextArea/blob/d84a498dd8c2e042ced9ccde4c96ea1d54770f68/src/main/java/org/fife/ui/rtextarea/RTADefaultInputMap.java where, effectively, they are using this KeyStroke for delete the rest of line 
```

public RTADefaultInputMap() {

		super();

		int defaultModifier = getDefaultModifier();
		//int ctrl = InputEvent.CTRL_MASK;
		int alt = InputEvent.ALT_MASK;
		int shift = InputEvent.SHIFT_MASK;
		boolean isOSX = RTextArea.isOSX();
		int moveByWordMod = isOSX ? alt : defaultModifier;

		put(KeyStroke.getKeyStroke(KeyEvent.VK_HOME,   0),					isOSX ? DefaultEditorKit.beginAction : DefaultEditorKit.beginLineAction);
                ...
                ...
                ...
		put(KeyStroke.getKeyStroke(KeyEvent.VK_DELETE, defaultModifier),			RTextAreaEditorKit.rtaDeleteRestOfLineAction);
                ...
                ...
}
		
```

[RTADefaultInputMap](https://github.com/bobbylight/RSyntaxTextArea/blob/d84a498dd8c2e042ced9ccde4c96ea1d54770f68/src/main/java/org/fife/ui/rtextarea/RTADefaultInputMap.java)

I was trying to add a new entry to the InputMap but there is not an action that deletes the next word.
Something like this:

```
private void ctrlDelete(){
    KeyStroke keyStroke = KeyStroke.getKeyStroke(KeyEvent.VK_DELETE, Toolkit.getDefaultToolkit().getMenuShortcutKeyMask());
    super.getInputMap().put(keyStroke, DefaultEditorKit.deleteNextWordAction);
  }
```

I don't know if there is an easier way to do that, but I got blocked there.

dtwaddell, 2016-04-04:
I notice in the beginning comments section in the same code you just quoted, the author had this to say:

> /**
 // The default input map for an <code>RTextArea</code>.  For the most part it is
 // exactly that the one for a <code>JTextArea</code>, but it adds a few things.
 // Currently, the new key bindings include:
 // 
 //   HOME key toggles between first character on line and first non-whitespace character on line.
 //   INSERT key toggles between insert and overwrite modes.
 //   Ctrl+DELETE key deletes all text between the caret and the end of the current line.
 //   Ctrl+Shift+Up and Ctrl+Shift+Down move the current line up and down, respectively.
 //   Ctrl+J joins lines.
 //   Ctrl+Z is undo and Ctrl+Y is redo.
 //   Ctrl+Up and Ctrl+Down shift the visible area of the text area up and down one line, respectively.
 //   F2 and Shift+F2 moves to the next and previous bookmarks, respectively.
 //   Ctrl+F2 toggles whether a bookmark is on the current line etc.
 
Apparently it was intentional?  Does that mean if you remove the explicit handler for VK_DELETE with defaultModifier, it will default to normal behavior (delete only one word)?

avargas-nearsoft, 2016-04-04:
Apparently I cannot directly remove an entry from the InputMap. Instead of that, I re-put the key with a new action:

```
private void fixCtrlDeleteBehavior() {
    KeyStroke keyStroke = KeyStroke.getKeyStroke(KeyEvent.VK_DELETE, Toolkit.getDefaultToolkit().getMenuShortcutKeyMask());
    getInputMap().put(keyStroke, SketchTextAreaEditorKit.rtaDeleteNextWordAction);
  }
```

It updates the current value for that KeyStroke.
I am going to send the pull request for those changes.

dtwaddell, 2016-04-04:
@avargas-nearsoft That's great.  Does that action for delete next word work?

cmaglie, 2016-04-07:
Fixed by #4815

closed 2016-04-07 by cmaglie