arduino/Arduino #4779: Add support for Let's Encrypt SSL certificate in bundled JRE cacerts file

Micmac2, 2016-03-30:
Currently, the board manager is unable to download JSON file stored on an HTTPS website that use Let's Encrypt Certificate.
It would be nice if the JRE bundled with the Arduino IDE could add support for the required certificates in its Java cacerts file.
Here is how to make it work on Mac OS X once the version 1.6.6 is installed (in a Terminal) :
wget https://letsencrypt.org/certs/isrgrootx1.pem
wget https://letsencrypt.org/certs/letsencryptauthorityx1.der
sudo keytool -trustcacerts -keystore '/Applications/Arduino 1.6.6.app/Contents/PlugIns/JavaAppletPlugin.plugin/Contents/Home/lib/security/cacerts' -storepass changeit -noprompt -importcert -alias letsencryptauthorityx1 -file ./letsencryptauthorityx1.der
sudo keytool -trustcacerts -keystore '/Applications/Arduino 1.6.6.app/Contents/PlugIns/JavaAppletPlugin.plugin/Contents/Home/lib/security/cacerts' -storepass changeit -noprompt -importcert -alias isrgrootx1 -file ./isrgrootx1.pem

per1234, 2016-03-30:
Does this issue cause the `file signature verification failed. File ignored.` warning when Boards Manager is opened? I was trying to help someone figure out why the JSON file stored on their website caused this warning but the same file on GitHub worked fine and we never found the cause.

Micmac2, 2016-03-31:
No, it is related to an error saying "PKIX path building failed" and "unable to find valid certification path to requested target" in old version of the Arduino IDE, or simply a simple error message at the bottom of the Board Manager Panel with the latest version.

facchinm, 2017-09-12:
Let's Encrypt root certificate should be included in JRE starting from 8u101 (http://www.oracle.com/technetwork/java/javase/8u101-relnotes-3021761.html), thus I'm marking this issue as fixed

closed 2017-09-12 by facchinm