arduino/Arduino #4783: [LibraryManager] Please add Yet Another PcInt library

paulo-raca, 2016-03-31:
https://github.com/paulo-raca/YetAnotherArduinoPcIntLibrary

Thanks

agdl, 2016-03-31:
Done thank you!

closed 2016-03-31 by agdl

NicoHood, 2016-03-31:
"Doesn't support user data on or pin state on callbacks"

I think there is an "on" too much.

I am not sure what this feature that is missing should be, but you might want something like this?
https://github.com/NicoHood/PinChangeInterrupt#get-trigger-on-mode-change

You made a very great comparison though. I know my code is messy, but I've no idea how to optimize it like this (low and high level API) with less definitions.

paulo-raca, 2016-03-31:
Yes, that is the feature I meant. I've already fixed the comparisson table.

Your code is "messy" because you've made different trade-offs:
- Your code is very efficient and configurable, but it  comes with extra complexity.
- Mine is simpler, but wastes more RAM and CPU cycles with pins that are not used

Thanks for taking a look :D

NicoHood, 2016-03-31:
Sure. Thats all great. Good to see some variations :)