arduino/Arduino #4784: ATtiny Compile Fail: Redefinition of preproc. symbol "BIN" (from "Stream.h" and "Print.h") by "iotnx4.h"

pediRAM, 2016-03-31:
If you **try to compile the code below for an ATtiny 84**, you will get the error messages after the code.
**The preprocessor symbol "BIN" used in "Stream.h" and "Print.h" (line 32, value=2) is redefined by "iotnx4.h" (line 71, value=7).** Because "iotn84.h" includes "iotnx4.h" (line 38) the value of "2" is overwritten by "7"!
**Quck-Fix / Workaround:** if you don't use "BIN" in "print" or "println" statements like ("print(var, BIN)"), you can comment line 32 in "Print.h" to get rid of the error.
```

#define PIN_MOTD_IN 0
#define PIN_LED_OUT 1

int val = 0;

void setup()
{
    pinMode(PIN_MOTD_IN, INPUT);
    pinMode(PIN_LED_OUT, OUTPUT);
    digitalWrite(PIN_LED_OUT, LOW);
}

void loop()
{
    val = digitalRead(PIN_MOTD_IN);
    digitalWrite(PIN_LED_OUT, val);
    delay(50);
}

```
_START_OF_ERROR_OUTPUT -->_
```
START_OF_CITATE -->
In file included from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Stream.h:26:0,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\HardwareSerial.h:29,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Arduino.h:224,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\USBAPI.h:33,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\CDC.cpp:19:

C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Print.h:32:0: warning: "BIN" redefined [enabled by default]

 #define BIN 2

 ^

In file included from d:\program\arduino\hardware\tools\avr\avr\include\avr\iotn84.h:38:0,

                 from d:\program\arduino\hardware\tools\avr\avr\include\avr\io.h:440,

                 from d:\program\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:88,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\USBAPI.h:24,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\CDC.cpp:19:

d:\program\arduino\hardware\tools\avr\avr\include\avr\iotnx4.h:71:0: note: this is the location of the previous definition

 #define BIN     7

 ^

"D:\Programme\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny84 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_attiny -DARDUINO_ARCH_AVR   "-IC:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny14" "C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\HardwareSerial.cpp" -o "C:\Windows\TEMP\build5edc50d0570cc346ba257358f0f62ca3.tmp\core\HardwareSerial.cpp.o"
In file included from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Stream.h:26:0,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\HardwareSerial.h:29,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Arduino.h:224,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\HardwareSerial.cpp:29:

C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Print.h:32:0: warning: "BIN" redefined [enabled by default]
[... 317 lines cut ...]
                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\USBAPI.h:33,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\USBCore.cpp:19:

C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Print.h:32:0: warning: "BIN" redefined [enabled by default]

 #define BIN 2

 ^

In file included from d:\program\arduino\hardware\tools\avr\avr\include\avr\iotn84.h:38:0,

                 from d:\program\arduino\hardware\tools\avr\avr\include\avr\io.h:440,

                 from d:\program\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:88,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\USBAPI.h:24,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\USBCore.cpp:19:

d:\program\arduino\hardware\tools\avr\avr\include\avr\iotnx4.h:71:0: note: this is the location of the previous definition

 #define BIN     7
```
_<-- END_OF_ERROR_OUTPUT_

NicoHood, 2016-03-31:
I got the same error, but did not report it yet. I think there is almost no way to fix it instead of changing BIN to 7 and hope no other MCU defines this again. As it is used a kinda enum, there should be no problem in using number 7.

per1234, 2016-03-31:
Note that this is a warning, not an error and doesn't cause the compile to fail. I'm assuming you're using https://github.com/damellis/attiny but you'll also get this warning and https://github.com/arduino/Arduino/issues/4785 with Arduino Gemma compiling a blank sketch.

It seems like this warning could be fixed by changing cores\arduino\Print.h:32 to either:
```c
#undef BIN
#define BIN 2
```
or
```c
#ifndef BIN
#define BIN 2
#endif
```

NicoHood, 2016-03-31:
Getting rid of the warning isn't the final goal. This can still cause weird errors if sometimes BIN (depending on which headers are included) is 2 and sometimes 7. Then the print will not work. And therefor there is a warning.

I'd completely remove the BIN, make it depreciated and use something else. Also the BIN printing is almost useless to to very bad formatting. So one would rather use HEX (which still isnt perfect though). So in this case the 2nd solution would be better and at least keep the priority definition to the iotnx4 header. However I would still manually then insert a warning again, to remind the user of undefined behavior.

pediRAM, 2016-03-31:
@per1234 @NicoHood : thank you for your replies and your recommendations.
You're right. Removing "warning" output dosen't solve the issue... it may also hide real errors in future.
By the way I'm using standard Arduino IDE (not using "https://github.com/damellis/attiny").
Thank you!

NicoHood, 2016-03-31:
Oh I wouldnt close this issue. I'd just use the warning and correct it this way, that if a conflict happens, the iotnx4 header has priority. It would look like this (inside print.h)

```cpp
#ifndef BIN
#define BIN 2
#else
#warning Not using the correct BIN definition for print in this file! See #4784 on Github.
#endif
```

pediRAM, 2016-04-01:
@NicoHood Sorry for closing! I'm a github noob. 
I have re-opened the issue now.

NicoHood, 2016-04-01:
Please test and post the output (for tinx4)
https://github.com/arduino/Arduino/pull/4791
You might also want to check what is going on when using print on the x4 with BIN now.

You might also want to check if a normal compilation on an arduino uno works fine (without warning). Also test the printing of a variable via BIN). Its trivial, it should work though.

pediRAM, 2016-04-01:
@NicoHood As you recommended in your last two Posts... I've changed the "Print.h" as you mentioned before and tested this code first on Arduino/Genuino Uno (TEST_BIN_ON_ATTINY84 mus be commented), then tested it on ATtiny 84A (TEST_BIN_ON_ATTINY84 must be UNcommented).

IMHO, it would be better to rename the symbol in the files "iotn84.h" and "iotnx4.h", from "BIN" to something else (like "IOTNX4_BIN").

**The Change in "Print.h":**
Line 32
from (original): `#define BIN 2`
to (now):
```
// Special Attinyx4 definition conflict, give the internal iotnx4.h a higher priority
#ifndef BIN
#define BIN 2
#else
#warning Not using the correct BIN definition for print in this file! See #4784 on Github.
#endif
```

**Used Code for testing:**
```
/**
 * Symbol for testing BIN on ATtiny 84.
 * Comment the "#define" line for testing with
 * Arduino Uno.
 */
#define TEST_BIN_ON_ATTINY84

#ifdef TEST_BIN_ON_ATTINY84
    #include <SoftwareSerial.h>
    #define SERIAL_OBJ swSerial
#else
    #define SERIAL_OBJ Serial
#endif

// Digital Input Pin.
#define PIN_DIG_IN    0

// RS 232 -> RX Pin.
#define PIN_SW_SER_RX 2

// RS 232 -> TX Pin.
#define PIN_SW_SER_TX 3

#ifdef TEST_BIN_ON_ATTINY84
    // Serial object.
    SoftwareSerial SERIAL_OBJ(PIN_SW_SER_RX, PIN_SW_SER_TX);
#endif

int val = 0;

void setup()
{
    pinMode(PIN_DIG_IN, INPUT);
    
    SERIAL_OBJ.begin(9600);
    SERIAL_OBJ.println("Testing BIN symbol started...");
}

void loop() 
{
    val = digitalRead(PIN_DIG_IN);
    SERIAL_OBJ.println(val, BIN);
    delay(1000);
}
```

**Result of compilation for Arduino/Genuino Uno (Arduino IDE 1.6.7):**
No Errors, No Warinings.

**Result of compilation for ATtiny 84A (Arduino IDE 1.6.7):**
```
In file included from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Stream.h:26:0,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\HardwareSerial.h:29,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Arduino.h:224,

                 from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Tone.cpp:39:

C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Print.h:37:2: warning: #warning Not using the correct BIN definition for print in this file! See #4784 on Github. [-Wcpp]

 #warning Not using the correct BIN definition for print in this file! See #4784 on Github.

  ^

In file included from C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Tone.cpp:37:0:

C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Tone.cpp: In function 'void TIMER2_COMPA_vect()':

C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Tone.cpp:538:5: warning: 'TIMER2_COMPA_vect' appears to be a misspelled signal handler [enabled by default]

 ISR(TIMER2_COMPA_vect)

     ^
```

NicoHood, 2016-04-01:
Great, thx for the report.

Changing iotnx is impossible, as it is inside the avr library. avr-libc i think.

What does `SERIAL_OBJ.println(23, BIN);` output on the uno and especially on the 84? I think it might fail on the 84 (testing only 1 and 0 is not representative).

pediRAM, 2016-04-01:
> What does SERIAL_OBJ.println(23, BIN); output on the uno and especially on the 84?

ATtiny X4/X5 do not have hardware RS-232, so you have to include and use "SoftwareSerial.h". If you do this, you cannot use the standard "Serial" object (because the name is reserved by Arduino IDE), so you have to use other name like "mySerial" or in this case (my code above) "swSerial" to get rid of warnings/errors. I use the symbol "SERIAL_OBJ" to switch the name for µController architectures which does not have a hardware serial port.
if Arduino => `SERIAL_OBJ.println(23, BIN)` will be converted by Arduino IDE to `Serial.println(23, BIN)`, which will  output "00010111" on the serial client (PC).
if ATtiny X5/X4 => `Serial_OBJ.print(23, BIN)` will be converted by Arduino IDE to `swSerial.println(23, BIN)`, which will make the same output as above.

NicoHood, 2016-04-01:
I know what your codes does and what it should output. But have you tested it? I am not sure that the correct values will be printed.

agdl, 2016-07-11:
Attiny chips can be used with external cores like 
https://github.com/SpenceKonde/ATTinyCore
The only Arduino product based on these chips is the Gemma who has a tiny 85

NicoHood, 2016-07-11:
And thatswhy we need this fix. Please reopen.

zgoda, 2016-10-12:
I'm getting similar warnings when compiling for Digispark with ATTiny85:

```
/home/jazg/.platformio/packages/framework-arduinoavr/cores/digispark_tiny/Print.h:37:0: warning: "BIN" redefined [enabled by default]
#define BIN 2
^
/home/jazg/.platformio/packages/toolchain-atmelavr/avr/include/avr/iotnx5.h:55:0: note: this is the location of the previous definition
#define BIN     7
^
```

cmaglie, 2017-03-07:
In the current AVR core:

- `BIN` is being always redefined as `2` (therefore nobody used it to access the `BIN` bit of the `ADCSRB` register)
- a lot of people complained about the warnings
- nobody complained about malfunctions of the ADC

The solution proposed by @NicoHood in https://github.com/arduino/Arduino/issues/4791:

```c++
// Special Attinyx4 definition conflict, give the internal iotnx4.h a higher priority
#ifndef BIN
#define BIN 2
#else
#warning Not using the correct BIN definition for print in this file! See #4784 on Github.
#endif
```
- changes a long standing value of `BIN` from `2` to `7`
- doesn't remove the warning (instead make it persistent, since it can't be no more disabled neither with `-Wnone`)

IMHO it's better to keep the current value of BIN, even if it's wrong if used in the context of the `ADCSRB` register, something like:

```c++
#if defined(BIN)
#define AVR_BIN BIN
#undef BIN
#endif
#define BIN 2
```

matthijskooijman, 2017-03-07:
Agreed on the former part of the analysis, changing it is not a really good idea.

However, the proposed solution won't work, since preprocessor are evaluated lazily, in the sense that any macros inside a macro are only expanded *after expanding* the outer macro, not *when defining it*. E.g.:

```
matthijs@grubby:~$ cat foo.c
#define BIN 7

#if defined(BIN)
#define AVR_BIN BIN
#undef BIN
#endif

#define BIN 2

BIN
AVR_BIN
matthijs@grubby:~$ gcc -E foo.c
# 1 "foo.c"
# 1 "<built-in>"
# 1 "<command-line>"
# 31 "<command-line>"
# 1 "/usr/include/stdc-predef.h" 1 3 4
# 32 "<command-line>" 2
# 1 "foo.c"
# 10 "foo.c"
2
2
```

cmaglie, 2017-03-07:
Oh, I see, thanks for testing! then I'd remove the extra definition of `AVR_BIN` and just leave:

```C++
#if defined(BIN)
#undef BIN
#endif
#define BIN 2
```

commit c6bac064 (2017-03-08): AVR: Prevent warnings if BIN is previously defined

Fix #4784
Close #4791
closed 2017-03-08 by cmaglie