arduino/Arduino #4785: ATtiny compile fail: 'TIMER2_COMPA_vect' appears to be a misspelled signal handler

pediRAM, 2016-03-31:
Try to compile following code for an ATtiny 84A:
```
#define PIN_MOTD_IN 0
#define PIN_LED_OUT 1

int val = 0;

void setup()
{
    pinMode(PIN_MOTD_IN, INPUT);
    pinMode(PIN_LED_OUT, OUTPUT);
    digitalWrite(PIN_LED_OUT, LOW);
}

void loop() 
{
    val = digitalRead(PIN_MOTD_IN);
    digitalWrite(PIN_LED_OUT, val);    
    delay(50);
}
```
and you will get following warning:

```
rdware\avr\1.6.10\cores\arduino\Tone.cpp:37:0:

C:\Users\ED_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Tone.cpp: In function 'void TIMER2_COMPA_vect()':

C:\Users\Ed_Snowden\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino\Tone.cpp:538:5: warning: 'TIMER2_COMPA_vect' appears to be a misspelled signal handler [enabled by default]

 ISR(TIMER2_COMPA_vect)

     ^
```

agdl, 2016-07-11:
Attiny chips can be used with external cores like 
https://github.com/SpenceKonde/ATTinyCore
The only Arduino product based on these chips is the Gemma who has a tiny 85

closed 2016-07-11 by agdl

sonyhome, 2017-07-12:
Some comments on these issues:

ATtiny Compile Fail: Redefinition of preproc. symbol "BIN" (from "Stream.h" and "Print.h") by "iotnx4.h" #4784

-> Fix it in Print.h
* Either define an enum in print.h for DEC,BIN,OCT,HEXA
* Or rename them to DEC_NUMBER, BIN_NUMBER, etc.
* Or replace DEC by 10, and skip the define altogether (it's overkill anyways)
Only DEC is used, and has to be updated in 2 files.